realsenseai/realsense-ros #1485: Topic:/camera/depth/image_rect_raw

sayyidabeegam, 2020-11-03:
I saved the topics "/camera/color/image_raw" and "/camera/depth/image_rect_raw" to rosbag.
I checked the data available with each topic corresponding to a single image.
Topic "/camera/color/image_raw" having 92,1600 values corresponding to a single frame.
I understood that the value comes from  480 X 640 X 3, where  480 X 640 is image size and 3 for RGB channels.
Similarly "/camera/depth/image_rect_raw" having 614400 values.It comes from 480 X 640 X 2, but I don't get the meaning of 2.
Can you please help me by explaining what is 2 here indicates
Simply I am expecting the data decoding of topic "/camera/depth/image_rect_raw"

MartyG-RealSense, 2020-11-03:
Hi @sayyidabeegam   My research of your question found that the number on the end refers to how many bytes each pixel of the depth image requires.  So 640 x 480 x 2 = 614400 bytes.

sayyidabeegam, 2020-11-03:
So you mean the depth details of a single pixel is 2 byte information?

MartyG-RealSense, 2020-11-03:
I'm not sure it's so much about depth detail as the amount of bytes in total that an uncompressed image at a particular resolution uses.  The guide in the link below explains the subject well.

https://www.scantips.com/basics1d.html

MartyG-RealSense, 2020-11-11:
Hi @sayyidabeegam   Do you still require assistance with this case pleae?  Thanks!

MartyG-RealSense, 2020-11-19:
Case closed due to no further comments received.

closed 2020-11-19 by MartyG-RealSense

qgallouedec, 2021-03-05:
I found the solution. You probably decoded the message in `uint8` while it is encoded in `uint16`. The data of a message of the topic `/camera/depth/image_rect_raw` doesn't have 614400 values but 307200. 
So you get an image of size 480 X 640.

```python
flatten_depth_img = np.frombuffer(msg.data, dtype=np.uint16)  # shape =(width*height,)
depth_img = flatten_depth_img.reshape(msg.height, msg.width) # shape =(width, height)
```

ninafiona, 2021-03-05:
When I record data and view all my topics in rqt, the depth data type is displayed as unit8 and therefore when I save the data I get if of size 480x640x2. 
I save my data using: `rostopic echo -b ../file.bag -p /camera/depth/image_rect_raw > dpt.txt ` is there a way to save it correctly to 480x640x1?

qgallouedec, 2021-03-05:
I don't think it is possible : see the comments of this [answer](https://answers.ros.org/question/227295/mono16-images-displayed-as-8-bits-in-rostopic-echo/?answer=227298#post-id-227298).