realsenseai/realsense-ros #1486: kamalmehta22

kamalmehta22, 2020-11-03:
I am using realsense L515 Lidar camera. To get the organized point cloud in xyzrgb form, I was trying to use rs_rgbd.launch. 

**_When I run this command:_**  roslaunch realsense2_camera rs_rgbd.launch
**_I am getting this error._** 
kbm@ubuntu:~/catkin_ws$ roslaunch realsense2_camera rs_rgbd.launch
... logging to /home/kbm/.ros/log/dc4ebebe-1df2-11eb-8eb9-000c29f5a84e/roslaunch-ubuntu-11978.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: rgbd_launch
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/kbm/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

Although, this launch file is there in src folder, I do not know why I am not able to use it. Would you please help me with acquiring organized pointcloud in the form of xyzrgb? 

For referance: using rs_camera.launch, it works perfectly fine. However, the pointcloud is not organized in the way I want to use. 
I am using the latest version of ros wrapper and realsense SDK.

VamsiVangaveeti, 2020-11-04:
Hello!

I had the same problem, then I realized that in order to launch rs_rgbd.launch we need to install a rgbd-launch package.

Install the same using:
`sudo apt install ros-melodic-rgbd-launch`

and see if the error still persists. Installing this solved the problem for me, but I am not able visualize that rviz yet though.

Cheers!

RealSenseSupport, 2020-12-09:
Hi @kamalmehta22 

Were you able to get passed the issues you were facing?  Do you need anything further on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-03 by RealSenseSupport