realsenseai/realsense-ros #1487: Multiple Cameras: Only One Publishes

civerachb-cpr, 2020-11-03:
I'm working on a robot running ROS Melodic on Ubuntu 18.04 with 2 RealSense D435i cameras (one facing front, one facing rear).  We're seeing a persistent problem where one camera (usually the rear one) never publishes any data.  I'm not sure if this is a problem with librealsense2 or with the ROS driver, or with both.

First off, running rs-enumerate-devices simply says "No device detected. Is it plugged in?".  Both cameras are connected to USB3 ports:

```
$ lsusb
Bus 002 Device 005: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 002 Device 013: ID 8086:0b3a Intel Corp. 
Bus 002 Device 012: ID 8086:0b3a Intel Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 006: ID 046d:c21f Logitech, Inc. F710 Wireless Gamepad [XInput Mode]
Bus 001 Device 005: ID 8087:0aa7 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Output from dmesg after physically unplugging and reconnecting the cameras:

```
[  505.866514] usb 2-5.1: USB disconnect, device number 11
[  506.528175] usb 2-5.1: new SuperSpeed USB device number 12 using xhci_hcd
[  506.548784] usb 2-5.1: New USB device found, idVendor=8086, idProduct=0b3a
[  506.548789] usb 2-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  506.548793] usb 2-5.1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  506.548796] usb 2-5.1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  506.548799] usb 2-5.1: SerialNumber: <redacted>
[  506.550840] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  506.555841] uvcvideo: Unable to create debugfs 2-12 directory.
[  506.555983] uvcvideo 2-5.1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  506.556003] uvcvideo 2-5.1:1.0: Entity type for entity Processing 2 was not initialized!
[  506.556006] uvcvideo 2-5.1:1.0: Entity type for entity Camera 1 was not initialized!
[  506.556210] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.0/input/input24
[  506.556855] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  506.559870] uvcvideo: Unable to create debugfs 2-12 directory.
[  506.560009] uvcvideo 2-5.1:1.3: Entity type for entity Processing 7 was not initialized!
[  506.560015] uvcvideo 2-5.1:1.3: Entity type for entity Extension 8 was not initialized!
[  506.560019] uvcvideo 2-5.1:1.3: Entity type for entity Camera 6 was not initialized!
[  506.764071] hid-sensor-hub 0003:8086:0B3A.000C: No report with id 0xffffffff found
[  511.107531] usb 2-5.2: USB disconnect, device number 10
[  511.780107] usb 2-5.2: new SuperSpeed USB device number 13 using xhci_hcd
[  511.804850] usb 2-5.2: New USB device found, idVendor=8086, idProduct=0b3a
[  511.804855] usb 2-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  511.804858] usb 2-5.2: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  511.804862] usb 2-5.2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  511.804864] usb 2-5.2: SerialNumber: <redacted>
[  511.807570] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  511.812870] uvcvideo: Unable to create debugfs 2-13 directory.
[  511.813026] uvcvideo 2-5.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  511.813030] uvcvideo 2-5.2:1.0: Entity type for entity Processing 2 was not initialized!
[  511.813033] uvcvideo 2-5.2:1.0: Entity type for entity Camera 1 was not initialized!
[  511.813237] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.2/2-5.2:1.0/input/input25
[  511.813941] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  511.817467] uvcvideo: Unable to create debugfs 2-13 directory.
[  511.817627] uvcvideo 2-5.2:1.3: Entity type for entity Processing 7 was not initialized!
[  511.817631] uvcvideo 2-5.2:1.3: Entity type for entity Extension 8 was not initialized!
[  511.817635] uvcvideo 2-5.2:1.3: Entity type for entity Camera 6 was not initialized!
[  511.972100] hid-sensor-hub 0003:8086:0B3A.000D: No report with id 0xffffffff found
```

Launch files:

Front:
```
<launch>
  <arg name="serial_no"           default="<redacted>"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
[... 287 lines cut ...]
[ INFO] [1604436649.444171770]: Built with LibRealSense v2.38.1
[ INFO] [1604436649.444181220]: Running with LibRealSense v2.38.1
[ INFO] [1604436649.453342389]:  
[ INFO] [1604436649.457342394]: Setting spin rate to <300>
 03/11 15:50:51,587 ERROR [140364958656256] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
 03/11 15:50:51,587 ERROR [139891189933824] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
 03/11 15:50:51,588 WARNING [140365044782848] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1604436651.588187933]: Device 2/2 failed with exception: failed to set power state
[ERROR] [1604436651.588218108]: The requested device with serial number 021222072138 is NOT found. Will Try again.
 03/11 15:50:51,588 WARNING [139891223504640] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1604436651.588239416]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1604436651.588262802]: The requested device with serial number 021122070743 is NOT found. Will Try again.
[ WARN] [1604436657.594595215]: No RealSense devices were found!
[ WARN] [1604436657.594697188]: No RealSense devices were found!
[ WARN] [1604436663.601063078]: No RealSense devices were found!
...
```

Output from starting ROS after rebooting: 
[ros-output-no-frames.txt](https://github.com/IntelRealSense/realsense-ros/files/5484047/ros-output-no-frames.txt)

Output from starting ROS after rebooting with only 1 camera connected; the front camera has been physically unplugged:
[ros-output-one-camera.txt](https://github.com/IntelRealSense/realsense-ros/files/5484065/ros-output-one-camera.txt)

Why are my cameras only being detected intermittently, and why does unplugging one of them seem to fix the other?  And what do I need to do to get both cameras to work all the time?

MartyG-RealSense, 2020-11-04:
Hi @civerachb-cpr   Which computer are you using the cameras with, please?  The behavior that you describe (getting a **Failed to set power state** with more than one camera attached but one camera alone works) reminds me of multicam behavior reported in the past on Nvidia Jetson boards.

civerachb-cpr, 2020-11-04:
The computer is an ASROCK Z390M-ITX with an Intel CPU.  It's an i9 or i7; I don't remember exactly which and the robot is powered-down with me working remotely at this exact moment.  If you need that level of information I can get someone to check the robot for me.

MartyG-RealSense, 2020-11-04:
The board's specification looks more than powerful enough to provide excellent performance with RealSense. 

If one camera works and the other does not, and each camera can work okay if used on its own, then I suspect that the problem may be that the serial number details necessary for ROS to identify the cameras as two separate cameras may not be set up, and that once it detects one camera then it ignores the other camera.

I see that you have defined two separate `<launch> </launch>` sections for front and rear cameras.  If the **front** launch sequence is run first and always works then the **rear** sequence is perhaps ignored. 

My understanding is that if your launch file is based on **rs_camera** then each camera should be launched from a separate terminal.  Otherwise, you can use **rs_multiple_devices.launch** and define the serial numbers for all of the cameras in a single roslaunch instruction.  This subject is discussed in the section of the RealSense ROs wrapper documentation in the link below:

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

Intel also has a RealSense ROS tutorial for setting up two 400 Series cameras on the same computer in two separate terminals that use **roslaunch realsense2_camera rs_camera.launch**.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

civerachb-cpr, 2020-11-04:
I just tried the rs_multiple_devices launch file and it also fails to launch both cameras:

```
$ roslaunch realsense2_camera rs_multiple_devices.launch 
... logging to /home/administrator/.ros/log/ae8747ec-1ed0-11eb-bbce-7085c2d586bb/roslaunch-cpr-a200-0626-2165.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cpr-a200-0626:38675/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 0
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 0
 * /camera1/realsense2_camera/gyro_fps: 0
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera1_infra1_op...
 * /camera1/realsense2_camera/infra2_frame_id: camera1_infra2_frame
[... 237 lines cut ...]
 04/11 14:05:24,314 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,357 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,371 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,415 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,429 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,472 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,486 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,530 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,544 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,587 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 04/11 14:05:24,831 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/11 14:05:24,875 WARNING [139765953328896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

The above test was done with the RealSense nodes being the only ROS nodes running on the robot.

Similarly, running rs_camera.launch from two separate logins (the robot is headless, so SSH'ing in from two different terminals) doesn't reliably work either.

MartyG-RealSense, 2020-11-05:
I note that on your multiple devices roslaunch instruction you did not include the serial numbers of the two cameras in the launch statement by defining serial_no_camera1 and serial_no_camera2.

**roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=<serial number of the first camera> serial_no_camera2:=<serial number of the second camera>**

civerachb-cpr, 2020-11-05:
The result is the same if I specify the serial numbers.  Omitting them causes the ROS node to auto-detect the serial numbers, which it appears to do correctly.

Output when specifying both serial numbers:

```
$ roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=021122070743 serial_no_camera_2:=021222072138
... logging to /home/administrator/.ros/log/6c50ec54-1fa0-11eb-9ef0-7085c2d586bb/roslaunch-cpr-a200-0626-1918.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cpr-a200-0626:41637/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 0
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 0
 * /camera1/realsense2_camera/gyro_fps: 0
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
[... 322 lines cut ...]
 05/11 14:52:31,421 WARNING [140292111968000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 14:52:31,472 WARNING [140292111968000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 14:52:33,533 WARNING [140292111968000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1604605956.962807685]:  
[ INFO] [1604605957.072255250]: Device with serial number 021222072138 was found.

[ INFO] [1604605957.072295496]: Device with physical ID 2-5.1-3 was found.
[ INFO] [1604605957.072310626]: Device with name Intel RealSense D435I was found.
[ INFO] [1604605957.072704834]: Device with port number 2-5.1 was found.
 05/11 14:52:37,082 ERROR [139659870967552] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 05/11 14:52:37,082 WARNING [139660040140544] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1604605957.082536395]: Device 2/2 failed with exception: failed to set power state
[ERROR] [1604605957.082549455]: The requested device with serial number 021122070743 is NOT found. Will Try again.
 05/11 14:52:39,589 WARNING [140292111968000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1604605963.095827948]:  
[ INFO] [1604605963.198989359]: Device with serial number 021222072138 was found.

[ INFO] [1604605963.199033754]: Device with physical ID 2-5.1-3 was found.
[... cut ...]
[ INFO] [1604605963.199480605]: Device with port number 2-5.1 was found.
 05/11 14:52:43,209 ERROR [139659887752960] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 05/11 14:52:43,209 WARNING [139660040140544] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1604605963.209540362]: Device 2/2 failed with exception: failed to set power state
[ERROR] [1604605963.209554447]: The requested device with serial number 021122070743 is NOT found. Will Try again.
```

MartyG-RealSense, 2020-11-06:
Further research found a past ROS case where a range of computers were tested with two D435 cameras (mostly Intel NUC computers and one ASROCK) and the ASROCK was the one that gave the most problems.  They list their results in a table at the bottom of the opening comment.

https://github.com/IntelRealSense/librealsense/issues/1615

civerachb-cpr, 2020-11-06:
Okay. So is there an actual solution to this problem?  Or are you just saying "We know the cameras don't reliably work in some computers. Sorry."

MartyG-RealSense, 2020-11-06:
I was highlighting that there is a performance precedent with the type of hardware that you are using.  There are some instances where a problem may not be easily solveable because of hardware design decisions that are outside of the control of RealSense.

The main problem seems to be that **rs-enumerate-devices** was stating that the cameras could not be detected, according to your opening comment.  Applying a method from problems with multiple cameras on Jetson to your ASRock system: is there any change is success of detection if you unplug both cameras, and then plug them in and check them with rs-enumerate devices.  First plug in Front first and Rear second.   If that does not work, unplug the cameras and this time plug Rear in first and Front second.  

The rationale behind this was that with Jetson, the order in which cameras are inserted could influence whether they were detected successfully or had a **Failed to set power state** error.

doronhi, 2020-11-08:
Hi @civerachb-cpr,
It seems like an actual issue with USB controller – in most cases there is no recovery, only restart. In severe cases you’ll also have to replug the camera.
Connecting the cameras using a powered external usb-hub may help. Is that an option for you?

civerachb-cpr, 2020-11-09:
I'm not sure if we have room inside this robot for a powered hub.  I suspect not.  But it's good to know in-future that we may need to budget the space for the additional hardware.

Thanks!

closed 2020-11-09 by civerachb-cpr

doronhi, 2020-11-10:
I could also suggest using a PCI-E extender as an alternative, to bypass the current USB controller.