realsenseai/realsense-ros #1490: How to get 16 bit depth image from L515 into ROS and then how to convert it into real world units

kashishdhal, 2020-11-06:
Hello,

I have connected the L515 Lidar camera with my computer over the ROS network. I am getting an image from the ROS topic "/camera/depth/image_rect_raw" which is of the type "sensor_msgs/Image".

If I look at one of the topics in MATLAB it looks like this:

MessageType: 'sensor_msgs/Image'
Header: [1×1 Header]
Height: 480
Width: 640
Encoding: '16UC1'
IsBigendian: 0
Step: 1280
Data: [614400×1 uint8]

Now I take this msg and convert it into two pictures in the following manner:

imgData = msg.Data;
imgH = msg.Height;
imgW = msg.Width;

%Create blank image output with required size
img = zeros(msg.Width,msg.Height,2);
imgD = reshape(imgData(1:2:end),imgW,imgH)';
imgC = reshape(imgData(2:2:end),imgW,imgH)'; 

This is how my depth image imgD looks:
[image]

This is how imgC looks:
[image]

Now my question is why am I getting 8 bit depth image? Shouldn't it be 16 bit? Why am I getting 480x640 pixels? Secondly, how do I convert the value of the pixels from the 0-255 to meters.

Please let me know if I am doing something wrong.

Thanks for the help in advance.!!

Regards,
Kashish Dhal

RealSenseSupport, 2020-12-17:
Hi @kashishdhal 

With the latest FW and SDK release are you still seeing this issue?  Have you tried looking at images outside of the ROS and Matlab environment?

kashishdhal, 2020-12-17:
Hello @RealSenseSupport ,

It's been a while since I posted this issue, it has been resolved now! Thanks!

Users reading this post may refer to this:
https://support.intelrealsense.com/hc/en-us/community/posts/360052364493-How-to-convert-the-ROS-Image-obtained-from-the-L525-Lidar-camera-to-real-world-units

closed 2020-12-17 by kashishdhal