realsenseai/realsense-ros #1491: URDF and Meshes for D455?

sisaha9, 2020-11-09:
Where can we find the URDF and Meshes for Intel Realsense D455 camera?

MartyG-RealSense, 2020-11-09:
Hi @sisaha9   The **URDF** and the **Meshes** folders of the RealSense ROS wrapper do not currently list files for the D455.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_description

The official RealSense CAD file archive lists a STEP format model for the D455 though if that is of any help.

https://dev.intelrealsense.com/docs/cad-files

MartyG-RealSense, 2020-11-17:
Hi @sisaha9   Do you still require assistance with this case, please?   Thanks!

MikeHallettUK, 2020-11-18:
Yes please ... I also have a D455 ... do we just copy / rename the D435i and IMU .xacro files ?

Or can you please add these D455 equivalents into realsense-ros/realsense2_description/urdf/   ?

Thanks

MartyG-RealSense, 2020-11-18:
Hi @MikeHallettUK  This request may be best handled as a **Pull Request** on this GitHub to add D455 urdf support, especially as @doronhi the RealSense ROS wrapper developer has just added an urdf PR for the L515 camera model.

https://github.com/IntelRealSense/realsense-ros/pulls

MartyG-RealSense, 2020-11-26:
Hi @sisaha9  Do you require further assistance with this case, please?   Thanks!

MikeHallettUK, 2020-11-26:
No thanks ... I did do this as a Pull Request.  Thanks

MartyG-RealSense, 2020-11-26:
Thanks @MikeHallettUK   The question was to @sisaha9 as they created this discussion.  Thanks very much for your pull   :)

sisaha9, 2020-11-28:
Yup we can close it. Sorry for the delay and thanks!

closed 2020-11-28 by sisaha9

MartyG-RealSense, 2020-11-28:
Thanks very much for the update!

doronhi, 2020-11-29:
@MikeHallettUK 

> No thanks ... I did do this as a Pull Request. Thanks

Hi @MikeHallettUK,
I'm sorry if I understand your meaning incorrectly, but did you submit a PR? I couldn't find any by your name. If you did, could you share a link please? If not, I'm sorry and please ignore this.

MikeHallettUK, 2020-11-29:
As MartyG suggested ... I added a comment to "Add urdf file for l515" = Can you please also add the D455 equivalents (like D435i and IMU .xacro files ?) into realsense-ros/realsense2_description/urdf/ ? @ [https://github.com/IntelRealSense/realsense-ros/pull/1508](https://github.com/IntelRealSense/realsense-ros/pull/1508)

doronhi, 2020-11-29:
Right. I remember now. I have no news yet except that it is already on my list.

felixvd, 2021-03-02:
Any chance we'll get the URDF and D455 meshes in this package soon? Friendly ping @dorodnic

MartyG-RealSense, 2021-03-02:
Friendly ping to @doronhi about the above question   :)

MikeHallettUK, 2022-07-09:
Got the URDF and D455 meshes now in new release THANKS.

... but I now "see" the D455 mesh visualisation appearing in the RVIZ camera view (superimposed upon the video feed): this appears intermittently when the robot uses movebase and RTABmap odom<>map transforms (which "jerk" around a bit with RTAB continual corrections to the tf).

I tried changing the <visual>, setting the mesh a little further back: this moved the mesh in the ROBOT-Model, but did not fix this irritating effect.

Deleting the <visual> section did fix the problem.... but obviously, also removed the mesh from the ROBOT-Model.

No show stopper... but, curious.

Mike