realsenseai/realsense-ros #1492: Ros IMU package for d455

siddharthcb, 2020-11-10:
hi,
i wanted to try out imu accuracy of d455 in ros for my robot.
when i launch this command: roslaunch realsense2_camera rs_camera.launch

there is no odom/gyro/accel topics being published as seen below.

$ rostopic list

/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

what shoud i be doing to get the imu in ros?

MartyG-RealSense, 2020-11-10:
Hi @siddharthcb   The gyro and accel topics are disabled by default in the RealSense ROS wrapper and have to be enabled at launch.  A way to do this is to add the commands below to the end of your roslaunch instruction:

**enable_gyro:=true enable_accel:=true**

siddharthcb, 2020-11-12:
hi @MartyG-RealSense ,
doesn`t d455 publish odom topic that is seen in t265?

MartyG-RealSense, 2020-11-12:
I do not have a definite answer for this question.  Generally, the odom topics seem to be most commonly used with T265, though I do not have information about whether they are exclusive to T265 or can work with other RealSense models with an IMU such as D435i and D455.

However, some people have succeeded in accessing odometry with the D435 / i models using **RTABMAP**.  Here are a couple of example links:

https://www.reddit.com/r/ROS/comments/hbd15y/rtabmap_ros_melodic_with_intel_realsense_d435_and/

https://answers.ros.org/question/317419/realsense-d435-rtabmap-pixhawk-imu-robot_localization/

siddharthcb, 2020-11-16:
Thanks @MartyG-RealSense. i will look into it

closed 2020-11-16 by siddharthcb