realsenseai/realsense-ros #1493: (Depth, 0) sensor isn't supported by current device! -- Skipping...

hashim19, 2020-11-10:
I tried running the t265 tracker with ubuntu 18 and ROS melodic, but I get this warning "(Depth, 0) sensor isn't supported by current device! -- Skipping..." and there are no topics such as, 

/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra2/camera_info
/camera/infra2/image_rect_raw

I went through [this]( #https://github.com/IntelRealSense/realsense-ros/issues/1191) whole thread but I could not solve my problem. 

Here is the complete log when I run "roslaunch realsense2_camera rs_t265.launch", 

[realsense_error_logs.txt](https://github.com/IntelRealSense/realsense-ros/files/5516591/realsense_error_logs.txt)

What should I do in this case? 

Besr Regards, 
Hashim Ali

hashim19, 2020-11-10:
It seems like this issue is being solved only by transitioning to ubuntu 16 and ROS kinetic.

doronhi, 2020-11-10:
T265 does not provide depth or color images. It can provide fisheye images if you set them ('enable_fisheye0:=true').
What I can't understand is how you got these topics on your Ubuntu16...
To be clear, there is no difference, that I am aware of, between the versions running on Kinetic and Melodic.

hashim19, 2020-11-10:
Ahhh.... okay. I get it now. Btw I am running ubuntu 18.

Thank you for your answer. You can close this issue.

Best,

closed 2020-11-10 by hashim19