realsenseai/realsense-ros #1494: Supported ROS Distributions

ranjitkathiriya, 2020-11-10:
Hey there,

In the real-sense package is also providing support with new ros noetic version as well?

If Yes, Can you please provide some tutorials, documentation that I can look at before starting the integration? I have never integrated real sense with ros so.

Thanks,

MartyG-RealSense, 2020-11-10:
Hi @ranjitkathiriya  A Noetic version of the RealSense ROS wrapper is planned for December 2020 if all goes well, according to **doronhi** the wrapper developer.

https://github.com/IntelRealSense/realsense-ros/issues/1423#issuecomment-702108336

In the meantime, some RealSense ROS users have succeeded in building the wrapper from **source code** on Noetic.  For example, the link below contains an installation guide from somebody who did so on Ubuntu 20.04.

https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/

ranjitkathiriya, 2020-11-11:
Thanks for helping me out

closed 2020-11-11 by ranjitkathiriya

MartyG-RealSense, 2020-11-11:
You are very wecome   :)

ranjitkathiriya, 2020-11-19:
Hi Marty,

I have followed these steps everything working perfectly. Even I am able to see the video in rviz.
.
https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/

But I am not able to get D435i No PointCloud2 for that.

[image]
[image]

Thanks