realsenseai/realsense-ros #1495: Save/Extract RGBD Frames

hosseinshirali, 2020-11-11:
Hello
I am new to this field and now i am able to get 3D point cloud of the environment by using Realsense D455 but I also want to save/extract the RGB and depth frames,  how is it possible to do that?
besides, I want to link each point clouds to its own frame, I mean I want to get a portion of point clouds then find that frames correspond to that region of point clouds.
What is a possible way to achieve that?

Thanks

MartyG-RealSense, 2020-11-11:
Hi @hosseinshirali   Do you want to use ROS to achieve these goals, or are you happy to use any method?  Thank you very much.

hosseinshirali, 2020-11-11:
hi @MartyG-RealSense, thanks for your quick response
for the SLAM i use ROS but for the solutions, I am open to using any method
thanks

MartyG-RealSense, 2020-11-11:
In regard to the image capture part of your question: outside of ROS, you could extract RGB and depth frames using the built-in snapshot feature in the **RealSense Viewer** program.  When a stream is active, it is a camera-shaped icon at the top of the stream panel.

[image]
 
Using this option on the Depth stream exports three files with depth data: an image PNG, a csv and a .raw (a type of image that preserves depth information).

[image]

Using the snapshot option on the RGB stream exports one PNG image file.

[image]

MartyG-RealSense, 2020-11-11:
In regard to the point cloud comparison aspect of your question: as you would likely wish to perform that in ROS too if possible, doing **correspondence** analysis on point clouds in ROS may suit your needs.

http://wiki.ros.org/pointcloud_registration

hosseinshirali, 2020-11-11:
@MartyG-RealSense  thank you
i think it is better to explain more, actually, I do the SLAM in ROS and I want to save the frames when I am creating the 3Dpoint clouds of the environment from that frames and also link points and frames together 
so realsense-viewer is not useful because I need too many frames
for ROS, is it possible to modify the source code to save the RGBD frames and also link the points and frames together without any other action?
if it is not possible or it is not the best way, i think, save data as rosbag file and extract frames from rosbag file are useful but pointcloud_registration is not useful because i already have a global pointcloud 
do you have any other solution for that? 
any hint would be a great help for me
thanks

MartyG-RealSense, 2020-11-12:
Hi @hosseinshirali   Apologies for the delay in responding further, as I was carefully considering and researching your question.

When you ask about linking points and frames together in ROS, i wonder if the information in the link below fits what you have in mind.

https://answers.ros.org/question/35084/transform-a-point-from-pointcloud/

In regard to saving RGBD frames: the **image_view** ROS package has **image_saver** and **extract_images** tools that work very similarly to each other, according to the documentation in the link below:

http://wiki.ros.org/image_view#image_view.2Fdiamondback.image_saver

hosseinshirali, 2020-11-13:
thank you @MartyG-RealSense 
with image_saver I am able to save RGB and depth frames separately also I can save them in rosbag file but I am not able to extract them from this rosbag file(i used bag_tools)
I would be grateful if you could help me with that
thanks

MartyG-RealSense, 2020-11-13:
Documentation for extracting images from a rosbag in ROS can be found here:

http://wiki.ros.org/rosbag/Tutorials/Exporting%20image%20and%20video%20data

hosseinshirali, 2020-11-13:
@MartyG-RealSense thank you
now i am able to extract rgb and depth frames but i have another problem
i saved these topics in rosbagfile :
topics:      /camera/color/camera_info            441 msgs    : sensor_msgs/CameraInfo      
             /camera/color/image_raw              441 msgs    : sensor_msgs/Image           
             /camera/depth/camera_info            338 msgs    : sensor_msgs/CameraInfo      
             /camera/depth/image_rect_raw         338 msgs    : sensor_msgs/Image           
             /camera/extrinsics/depth_to_color      1 msg     : realsense2_camera/Extrinsics
             /tf                                 5054 msgs    : tf2_msgs/TFMessage           (3 connections)
             /tf_static                             1 msg     : tf2_msgs/TFMessage
as you can see i saved 441 rgb frames and 338 depth frames
my problem is i want the same number of rgb and depth frames,why they are different?
the other thing is about when i extract them i got 426frames for rgb and 333 frames for depth, why i lost some frame?

MartyG-RealSense, 2020-11-13:
When using ROS to extract frames, there may be dropped frames if the images were compressed.

https://answers.ros.org/question/298180/extracting-compressed-image-data-from-bag-file-results-in-doubled-up-or-dropped-frames/

hosseinshirali, 2020-11-16:
@MartyG-RealSense thanks but they are not compressed
i used this command line for recording the topics (RGBD frames):
rosbag record /camera/aligned_depth_to_color/camera_info camera/aligned_depth_to_color/image_raw /camera/color/camera_info /camera/color/image_raw /camera/depth/camera_info /camera/depth/color/points /camera/depth/image_rect_raw /camera/extrinsics/depth_to_color /camera/imu /camera/imu_info /tf /tf_static -O temp_bag.bag
and the number of /camera/color/image_raw and /camera/depth/image_rect_raw are different 
i want one rgb frame and one depth frame for each timestamp
@doronhi can you help me with this problem?

doronhi, 2020-11-16:
There could be many reasons why the number of depth and color frames are not the same in the final rosbag file you saved. There could be temporal load on the system or on the disk for example. You could try and see if you get better results with lower resolution or frame rate but frame drops, from my experience, happen. 

The realsense2_camera's log could be useful to see what parameters you already use.
If I understand correctly, you don't set the enable_pointcloud parameter but constructs your own pointcloud. If that is the case, I would like to suggest using the enable_sync:=true parameter.
Depth and color are 2 separate streams. the enable_sync flag uses the librealsense2 mechanism unite frames with close timestamps into a frames set and then publishes them together **with the same timestamp**
It still does not guarantee you'll get an equal number of color and depth frames. If a color frame is dropped for any system reason, the synching algorithm wont wait and publish a frames set without the missing color frame. But it gives all the frames of a frame set the same timestamp. That way you could filter from the frames you saved the color frames without matching depth frames.

hosseinshirali, 2020-11-16:
@doronhi thank you, by changing the frame rate and also enable_sync:=true it gets better but again there is a small difference
I am new to this field would you please tell me how can I filter frames, as you told me?
my main purpose is to do object detection on pointcloud then do the same operation on related RGB frames...so I need to find a way that finds RGB frames that are connected to each point...
can you tell me is there any way to do that in ROS or if you know any other solution it will be a great help for me, thanks

doronhi, 2020-11-17:
I can't give you a code sample as I never done it. The "filtering" idea I had is to start reading the first rgb frame and the first depth frame. If they have the same timestamp - use them. If not, advance the pointer to the frame with the older timestamp and check again.

hosseinshirali, 2020-11-17:
@doronhi thank you 
Can you also help me with the second part of my problem?
matching 3d points to the RGB frames to do the same operation from point clouds to RGB frames

hosseinshirali, 2020-11-18:
@dorodnic i would be grateful if you could help me with this problem
thanks

MartyG-RealSense, 2020-11-19:
@hosseinshirali   Do you mean @doronhi  ?

Don't worry, I sometimes almost pick the wrong name in the tag selector too   :)

hosseinshirali, 2020-11-20:
@MartyG-RealSense yes i changed it, thanks

MartyG-RealSense, 2020-11-26:
Tagging a reminder for @doronhi about the question of @hosseinshirali in the link below   Thanks!

https://github.com/IntelRealSense/realsense-ros/issues/1495#issuecomment-728839609

doronhi, 2020-11-29:
Sorry for the late response. That is not something I did and I usually try to avoid recommending a path I didn't walk through myself.
I would try using the `rs2_project_point_to_pixel` function with the color-image intrinsics to get the 2D color-image coordinates from the 3D pointcloud coordinates. The included script [show_center_depth.py](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py) demonstrate the opposite conversion using `rs2_deproject_pixel_to_point` and will help you find out how to get the intrinsics needed for the conversion.

hosseinshirali, 2020-12-04:
thanks for your help, i will try this path

closed 2020-12-04 by hosseinshirali