realsenseai/realsense-ros #1497: problem with librealsense version check

mjsobrep, 2020-11-11:
I am having a problem with the version check from #1278. I am getting:

[ INFO] [1605119322.475257345]: RealSense ROS v2.2.18
[ INFO] [1605119322.475280472]: Built with LibRealSense v2.38.1
[ INFO] [1605119322.475454807]: Running with LibRealSense v2.39.0
[ WARN] [1605119322.475552420]: ***************************************************
[ WARN] [1605119322.475594147]: ** mjs: running with a different librealsense version **
[ WARN] [1605119322.475649595]: ** than the one the wrapper was compiled with!   **
[ WARN] [1605119322.475685700]: ***************************************************

However, I am building with 2.39.0. I have 2.39.0 installed (according to realsense viewer). I added a line to the cmake to print the version: `message(STATUS "Found LibRealSense Version: ${realsense2_VERSION}")` which prints: `-- Found LibRealSense Version: 2.39.0` at build time. And just to make sure that I am running and building what I think I am, I dropped my initials into the warning message in the source files, which prints out when running (seen above). Any thoughts on what may be causing this?

MartyG-RealSense, 2020-11-11:
Hi @mjsobrep   Did you build the 2.2.18 wrapper originally with SDK 2.38.1 and then upgrade librealsense to 2.39.0 at a later date after the wrapper had been built, please?  If you did, I would recommend rebuilding the ROS wrapper so that it builds on SDK 2.39.0.

mjsobrep, 2020-11-11:
I believe I have already done a clean build with 2.39.0, see above.

MartyG-RealSense, 2020-11-11:
Can you confirm that on your clean build, you built librealsense 2.39.0 first and then ROS wrapper 2.2.18 second please?

doronhi, 2020-11-11:
Could it be that you have realsense2_camera installed from debian package also? Try adding a message of your own to the realsense2_camera code to make sure you indeed running the app that you build.

mjsobrep, 2020-11-11:
@MartyG-RealSense I am not building librealsense, using the debian packages for that. 2.39.0 is installed prior to buidling the realsense-ros packages

@doronhi I don't think so. To check, I added my initials to the error printout (see first message). Is there another place that I should be adding a printout?

MartyG-RealSense, 2020-11-11:
@mjsobrep   Did you install the wrapper with packages using **Method 1** of the wrapper's installation instructions?

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

With this method, librealsense and the ROS wrapper should both be installed at the same time.  But the librealsense version in these packages may not be the latest one.  

So if you installed SDK 2.39.0 first with packages and then used wrapper installation Method 1, I could foresee how SDK 2.38.1 could get installed with the wrapper and you could end up with two SDK versions on your computer.

mjsobrep, 2020-11-11:
@MartyG-RealSense I use method 2. But one thing that comes to mind is that maybe when I use rosdep: `rosdep install --from-paths ~/catkin_ws/src/[...some package that depends on realsense...]/ --ignore-src -r -y` something gets off. However, Prior to building the realsense-ros package, if I run realsense-viewer, it says v2.39.0 and cmake reports that it is using v2.39.0 (as noted in the original post).

How is it possible for cmake to report one thing then at runtime for there to be another?

MartyG-RealSense, 2020-11-11:
In a past case, a RealSense team member suggested dealing with a ROS wrapper version mismatch by adding a "copy if newer" step to the pre-build process to guarantee the object file versions always match.

https://github.com/IntelRealSense/realsense-ros/issues/950

mjsobrep, 2020-11-11:
@MartyG-RealSense Thanks, you are right, there was another realsense version: `ros-melodic-librealsense2`. I'm pretty sure that I did not install that myself, so I am thinking that it came from rosdep seeing `  <exec_depend>realsense2_camera</exec_depend>
` in a package.xml. after uninstalling that, I am good. I am still not clear on why that was being used instead of the package that cmake was finding. Here is what I had installed before:

```

ii  librealsense2:amd64                          2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Cross Platform API - runtime
hi  librealsense2-dbg:amd64                      2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Camera Capture API - debug symbols
hi  librealsense2-dev:amd64                      2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Camera Capture API - development files
ii  librealsense2-dkms                           1.3.14-0ubuntu1                                  all          Modified kernel modules for librealsense2
ii  librealsense2-gl:amd64                       2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) - GLSL-enabled extensions
ii  librealsense2-net:amd64                      2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) - Network Bridge extension
ii  librealsense2-udev-rules:amd64               2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules
hi  librealsense2-utils:amd64                    2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Camera Capture API - utils and demos
ii  ros-melodic-librealsense2                    2.38.1-1bionic.20200915.132851                   amd64        Library for capturing data from the Intel(R) RealSense(TM) SR300, D400 Depth cameras and T2xx Tracking devices.

```
after uninstalling ros-melodic-librealsense2, everything works as expected.

MartyG-RealSense, 2020-11-11:
Great to hear @mjsobrep  - thanks for the update!

MartyG-RealSense, 2020-11-19:
Hi @mjsobrep   Is it okay to close this issue now please if you have had a satisfactory outcome?   Thanks!

closed 2020-11-29 by MartyG-RealSense