realsenseai/realsense-ros #1500: Release 2.2.19+

130s, 2020-11-13:
# Question
Is the release for 2.x planned this month November? If so is there an estimate of when that will be? Like the week of 15th? 22th? etc.

## Background of the question
Looking at [tags](https://github.com/IntelRealSense/realsense-ros/tags) page, almost every month a new release has been made for version 2.x. Frequent release is absolutely awesome! I think that gives the user land a lot of flexibility. Thank you for the attentive release maintenance.

That said, sometimes it's nice to know when the next release is planned to be made. At my team we lock down the version of software our application depends on and once we locked down, we won't typically update the version of upstream dependency easily. Now, however, because we've made some contributions recently to Realsense we're interested in updating the version to depend on.
Knowing the approximate timing of the new release (for 2.x for us) is helpful.

MartyG-RealSense, 2020-11-14:
Hi @130s   You can view which pulls have been merged into the ROS wrapper recently by visiting the **Commits** page of the RealSense ROS GitHub.  The contributions of @abhijitmajumdar and yourself re listed as having been committed on November 9 2020.

https://github.com/IntelRealSense/realsense-ros/commits/development

In regard to providing precise estimates of when new ROS wrapper versions will arrive: the librealsense SDK does not provide date-scheduled updates either, but rather will talk in terms of a new release arriving in more general terms such as "a couple of weeks" or "soon".   Or if a particular feature is planned for introduction further out, an approximate month such as 'December 2020'.  This scheduling flexibility is useful in software that has as many moving parts as the librealsense SDK and the ROS wrapper do.

@doronhi  Is the development branch of the RealSense ROS wrapper like the development branch of the main librealsense GitHub, where you can access in-development versions early with the **Code** button on the GitHub as changes get committed, please?

[image]

doronhi, 2020-11-15:
Hi,
@MartyG-RealSense , yes, in that sense, the development branch is like the librealsense2 development branch.
@130s , The last realsense2_camera was released to support the 2.39.0 version of librealsense2. A new librealsense2 version is expected to be released in the next couple of days and I'll try to follow it closely and release a new realsense2_camera shortly afterwards. It's very hard for me to commit to specific dates. I am sorry as I'm sure it causes much inconvenience for your team as well as many more users.

130s, 2020-11-16:
Thanks for clarification. I was just wondering if there's a room for providing more specific dates. Sounds like doing so is hard, which I totally understand.

I keep this ticket open as a "task to make new release". I've seen on many repos where community communicates with the maintainers about the upcoming releases.
If having a ticket like this is undesirable I'm fine in closing this one, as I got all I wanted to know.

130s, 2020-11-16:
Totally our ego, but if https://github.com/IntelRealSense/realsense-ros/pull/1504 can be merged and included in the next release, that'll be great.

MartyG-RealSense, 2020-11-23:
Hi @130s   Do you require further assistance with this case, please?   Thanks!

130s, 2020-11-23:
No, like I said in https://github.com/IntelRealSense/realsense-ros/issues/1500#issuecomment-728130189 I leave it up to you whether maintenance team wants to keep communicating about the next release on a ticket like this.

For this one particularly, I saw [2.2.20](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.20) was out a few days ago (Looks like 2.2.19 was skipped?), so the condition in the title statement "Release 2.2.19+" is met. I'm closing.

closed 2020-11-23 by 130s