realsenseai/realsense-ros #1501: allow for selecting cameras by ethernet instead of USB for ROS/ROS2 Drivers and Launch Files

wegunterjr, 2020-11-13:
I was pointed to an effort to get the Realsense 455 camera on ethernet instead of USB (cable management, industrial connections, etc, etc), and wondering if  it is possible to have the ROS driver look for the ethernet address instead of the USB Port?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

MartyG-RealSense, 2020-11-14:
Hi @wegunterjr   I researched your quetion and located a case where someone used ethernet with realsense2_camera by configuring **ROS_MASTER_URI**

https://answers.ros.org/question/357185/intel-realsense-nodelets-running-with-remote-roscore-are-dying-bond-broken/

doronhi, 2020-11-15:
Hi @wegunterjr ,
Currently it is not possible with realsense2_camera. I didn't try it yet but I can give you a starting point.
librealsense2 has a package called librealsense2-net. You can install it or build it from code using the BUILD_NETWORK_DEVICE flag in the cmake file. Best description of the package, it's interface and usage can be found in the [depth-camera-over-ethernet-whitepaper](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-sdk-integration).
realsense2_camera does not support this extension. Out of the top of my head, the gap is not so great. One will have to add a parameter for the IP-address to be used and update the context within realsense_node_factory.cpp accordingly.
It's currently not in my scope but if this direction is indeed useful to you, I think many will be happy to hear about your experience.

MartyG-RealSense, 2020-11-23:
Hi @wegunterjr   Do you require further assistance with this case, please?   Thanks!

wegunterjr, 2020-11-23:
I am finishing up my sprint but will spin this up again in a week!
Thanks for the suggestions!!!  I am excited to try them.

Walter Gunter
"The world has the habit of making room for the man whose actions show that
he knows where he is going.”  - Napoleon Hill

MartyG-RealSense, 2020-11-23:
Thanks so much for the update @wegunterjr  - good luck!

MartyG-RealSense, 2020-12-01:
Adding a note to keep this case open for a further time period.

MartyG-RealSense, 2020-12-08:
Hi @wegunterjr  Do you have an update about this case that you can provide, please?    Thanks!

wegunterjr, 2020-12-11:
not yet - got distracted with another part of the system.  Jumping back on this soon.

MartyG-RealSense, 2020-12-11:
Okay, thanks for the update!

wegunterjr, 2021-01-08:
Still using it in its default setup case.  Thinking of trying a couple of things today and this next week.  I will update as soon as I make progress.

MartyG-RealSense, 2021-01-08:
Thanks very much @wegunterjr for the update.   Good luck!

wegunterjr, 2021-01-25:
Ha..I literally just put it in the new sprint starting today.   Let me see
what I can do.

MartyG-RealSense, 2021-02-20:
Hi @wegunterjr Do you have an update about this case that you can provide, please?  This case will be closed in 7 days if we do not hear from you.  Thanks!

closed 2021-02-26 by MartyG-RealSense