realsenseai/realsense-ros #1503: Converting 2D image coordinates to 3D coordinates using ROS + Intel Realsense

ranjitkathiriya, 2020-11-16:
Hi there,

I am using Intel Realsense Depth camera 435i. Until now I have implemented an object detection code that gives me x,y coordinates, and depth (distance in meaters ). 
1 . Can I call depth which I am getting from this [link ](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb)as a Z coordinates? 
2. How can I convert these coordinates to 3d space so that my robot (Kuka-hand-eye robot) can know where to go for picking an object?  

Thanks for your time. once again Thanks

MartyG-RealSense, 2020-11-16:
A project in the link below called **AUBOi5-D435-ROS-DOPE** may be a useful reference for you. It uses RealSense D435, Pytorch and ROS to get the 6 degrees of freedom (6-DOF) pose of recognized objects for a robot-arm bin picking application.

https://github.com/avinashsen707/AUBOi5-D435-ROS-DOPE

ranjitkathiriya, 2020-11-17:
Thanks for the reply.  It is very helpful for me.

1. I cannot do with object detection? Like my question is I have x,y,z, coordinates of an object, and is there any way that we can do with this or we need object pose?

Thanks

MartyG-RealSense, 2020-11-17:
A problem with only having XYZ position coordinates is that it would make it difficult to work out how to handle the object with a robot arm if it was prone to having its angle changed (e.g if it was in a loose pile of parts in a picking bin).  Pose includes position and angle.

The official RealSense blog has an article about robot arm handling.

https://www.intelrealsense.com/robot-grasping/

ranjitkathiriya, 2020-11-19:
Thanks for help

closed 2020-11-19 by ranjitkathiriya