realsenseai/realsense-ros #1506: T265 and D455 are read, but data is wrong (Delayed by 5 seconds + )

AdamLink1270, 2020-11-17:
I am trying to get my T265 and D455 sensors working on a Raspberry Pi 4, running Ubuntu Sever 18.04 and Ros Melodic.

I have built librealsense from source, and the ros realsense node, I followed this guide for librealsense:
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

When I run the node my T265 is detected, and ros topics begin publishing- I don't get any errors there.

I start the node with roslaunch realsense2_camera rs_t265.launch

The topic publishes at a consistent 200hz, however the data is wrong- it will either simply not respond to me moving the sensor, or any movement (Looking at the odom/sample topic) will be published 5 seconds or later after the actual movement.
These delays effect the odom topic most severely, interestingly the accel topic is not affected- it responds to movements quickly. 

The sensor will always stop publishing after a few minutes (or even a few seconds), requiring being unpluged and plugged in again.

I get similar problems on the D455.

I have tested this sensor and proven its functionality on two other computers running Ubuntu 18, I did not build the libs myself there though- it ran perfectly on those computers.

I have made sure to remove the apt version of the ros realsense libraries, ensuring the version I built is being used.

I'm very confused as to what may be causing this, other ros nodes seem to publish fine

Any help is much appreciated

Schloern93, 2020-12-03:
I have the same error when I use the launch files.

snakehaihai, 2021-03-04:
also have similar issue.  
even with the same D435. the imu time are lagging behind the camera time.

closed 2026-10-23 by MartyG-RealSense