realsenseai/realsense-ros #1507: Error in upsuare bord 2

ranjitkathiriya, 2020-11-17:
17/11 17:07:06,311 ERROR [139856072140544] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 17/11 17:07:06,331 WARNING [139856167020288] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1605632826.331991814]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1605632826.332221769]: The requested device with  is NOT found. Will Try again.

Getting this error while running : roslaunch realsense2_camera rs_camera.launch

My platform is Ubuntu and its in up-square Bord 2
I am using Intel; realsense D435i

MartyG-RealSense, 2020-11-18:
Hi @ranjitkathiriya   The Up Squared board has been shown to be able to handle two RealSense 400 Series cameras that are plugged into it, so I would not expect to see a power problem with it when using one camera.

There was a recent ROS case involving this error where the cause was found to be to check whether a particular **udev rule** was installed, and to install it if it was not present.

https://github.com/IntelRealSense/realsense-ros/issues/1408#issuecomment-698128999

ranjitkathiriya, 2020-11-18:
Thanks for reply I have followed that steps and I got this error solved but I am getting this warning.

[ INFO] [1605695047.869371921]: insert Color to RGB Camera
 18/11 10:24:07,936 WARNING [139743807403776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1605695047.987951715]: SELECTED BASE:Depth, 0
[ INFO] [1605695048.020548037]: RealSense Node Is Up!
 18/11 10:24:08,089 WARNING [139743807403776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/11 10:24:08,140 WARNING [139743807403776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/11 10:24:08,191 WARNING [139743807403776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Thanks,
Ranjit

MartyG-RealSense, 2020-11-18:
If you only get several **control_transfer returned** warnings at launch and then they stop, or if you only receive one or two a minute after launch, then it is safe to ignore the warnings.  

If the launch process has not exited, I recommend checking RViz to see whether the topics are being published correctly despite the warnings.

closed 2020-11-20 by ranjitkathiriya