realsenseai/realsense-ros #1509: ERROR [5016] (types.h:307) get_device_time_ms() took too long

matejmeglic, 2020-11-19:
Hello.

Can somebody put me in the right direction as I get the error below:
- I use Windows 10
- installed ROS and ROS2, VS2019 Community
- D435i is updated to the latest version
- everything works in RealSenseViewer.exe
- when I try to connect it to **rwiz**, I can see img_raw and img_rect_raw, but nothing else

Thank you dearly. Matt

LOGS

`C:\catkin_ws>roslaunch realsense2_camera rs_camera.launch
... logging to C:\Users\matej.EXOTIC-HQ\.ros\log\40b24aad-29fa-11eb-bd05-48a472e65e55\roslaunch-CARBON-MATEJ-10556.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://127.0.0.1:52983/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
[... 75 lines cut ...]
[ INFO] [1605744042.184348900]: Stereo Module was found.
[ INFO] [1605744042.212817800]: RGB Camera was found.
[ INFO] [1605744042.213157000]: Motion Module was found.
[ INFO] [1605744042.213391100]: num_filters: 0
[ INFO] [1605744042.213784200]: Setting Dynamic reconfig parameters.
 19/11 01:00:43,241 WARNING [12556] (types.cpp:49) hwmon command 0x7d failed.
Error type: No data to return (-21).
hwmon command 0x7d failed.
Error type: No data to return (-21).
[ INFO] [1605744044.537149600]: Done Setting Dynamic reconfig parameters.
[ INFO] [1605744044.537772300]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1605744044.538442700]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1605744044.538514100]: setupPublishers...
[ INFO] [1605744044.541467100]: Expected frequency for depth = 30.00000
[ INFO] [1605744044.564727100]: Expected frequency for color = 30.00000
[ INFO] [1605744044.583148800]: setupStreams...
[ INFO] [1605744044.619664400]: insert Depth to Stereo Module
[ INFO] [1605744044.619915400]: insert Color to RGB Camera
[ INFO] [1605744044.716715600]: SELECTED BASE:Depth, 0
[ INFO] [1605744044.725328600]: RealSense Node Is Up!
 19/11 01:00:46,081 ERROR [5016] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 19/11 01:00:46,265 ERROR [5016] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
[ WARN] [1605744046.566843400]: Hardware Notification:Depth stream start failure,1.60574e+12,Error,Hardware Error
 19/11 01:00:46,707 ERROR [5016] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 19/11 01:00:58,911 ERROR [5016] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)`

MartyG-RealSense, 2020-11-19:
Hi @matejmeglic    I would not recommend using librealsense SDK **2.34.0** as it had a problem with a continuously generating **get_device_time_ms() took too long (more then 2 mSecs)** error.

 Also, the librealsense version that you use should be as close as possible to the version recommended by the release notes of a particular ROS wrapper version.  

https://github.com/IntelRealSense/realsense-ros/releases

For example, the **2.2.18** wrapper version that you are using is intended for use with librealsense SDK version **2.39.0**.

[image]

Regarding your statement that you installed ROS and ROS2: you would not be able to use the RealSense ROS wrappers with **ROS Wrapper 2.0** in their title with ROS2.  For ROS2, you should use the wrapper versions on the Releases page that have the name **ROS2-Eloquent Wrapper**.

matejmeglic, 2020-11-19:
Hey MartyG - I bow to you for a quick response, really appreciate that.

I will try uninstalling ROS2 and upgrading to LibRealSense v2.39.0 - any
hints on how to do that (sorry, that is PM trying to prototype).
I will poke around and try to find a solution myself.

Best, M.

Matej Meglič
matej@matejmeglic.com
+386 41 264 960

matejmeglic.com

MartyG-RealSense, 2020-11-19:
As you are using Windows 10, you can install SDK 2.39.0 quickly simply by downloading and running an installer file called **Intel.RealSense.SDK-WIN10-2.39.0.2337.exe** from the **Assets** file list at the bottom of the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.39.0

[image]

matejmeglic, 2020-11-23:
Hi Marty. 

I tried to install 2.39 and was successful (can see it in RealSenseViewer version), but for some reason, roslaunch still wants to launch with the old version. 

[image]

How can I uninstall LibRealSense 2.34 from Windows?
How can I set roslaunch to run with 2.39?

I tried rebuilding the whole thing, but it didn't work.

Thank you, Matej

MartyG-RealSense, 2020-11-23:
If you installed 2.34 with the installer file from the Assets list and installed 2.39 the same way then there would be no need to uninstall 2.34, as the newer SDK installation overwrites the previous one when installing with this method.

When librealsense is updated, you do need to rebuild the RealSense ROS wrapper though.  For SDK 2.39, you should build ROS wrapper **2.2.18** as the best match for 2.39.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.18

matejmeglic, 2020-11-23:
I installed 2.34 via .exe file weeks ago when I received a camera.

Steps:
- I deleted everything from catkin_ws.
- installed 2.39 via .exe file for Win10
- vcpkg install realsense2:x64-windows returns **Package realsense2:x64-windows is already installed**
- I followed https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources for rebuilding ROS wrapper
(I cloned **ddynamic_reconfigure** git repo into C:\catkin_ws\src\ and placed abovementioned 2.2.18 zip contents in the same folder)
- Successfully installed ROS
- devel\setup.bat
- roslaunch realsense2_camera rs_camera.launch <-- still runs LibRealSense 2.34

Do you have any other idea how to uninstall 2.34 ?

Thank you lots. M

MartyG-RealSense, 2020-11-23:
This reminds me of a past case involving a change of librealsense version by a RealSense ROS user.  I recommend reading the discussion from the point hat I have linked to below until the end of it.

https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581831329

In short though, the RealSense ROS user in that case deleted their workspace and then rebuilt it.

matejmeglic, 2020-11-23:
That is exactly what I did as well. 

Any clues on how to "clean" the system and start anew on windows (the case describe linux flow)?

Thank you.

MartyG-RealSense, 2020-11-23:
I am not sure why you would need to use **vcpkg install realsense2:x64-windows**.  The SDK should just automatically install on Windows 10 by running the **Intel.RealSense.SDK-WIN10-2.39.0.2337.exe** installer file from the Assets list without having to to input any other commands.

matejmeglic, 2020-11-24:
I decided to stop debugging this and migrated to Ubuntu VM. Installed everything with no problem. Closing - Marty's comment is likely correct and related.

closed 2020-11-24 by matejmeglic

MartyG-RealSense, 2020-11-25:
Thanks very much for the update @matejmeglic - great news that you succeeded!

majinshaoyuindustry, 2021-01-04:
> I am not sure why you would need to use **vcpkg install realsense2:x64-windows**. The SDK should just automatically install on Windows 10 by running the **Intel.RealSense.SDK-WIN10-2.39.0.2337.exe** installer file from the Assets list of 2.39.0 without having to to input any other commands.

I am running on exactly the same problem and I would like to point out that "vcpkg install realsense2:x64-windows" is what the instruction told us to do.

MartyG-RealSense, 2021-01-04:
Hi @majinshaoyuindustry   You are correct that the front page of librealsense does document a vcpkg installation method:

https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg

In practice vcpkg is rarely used for librealsense installation though, with the majority of Windows 10 users using the aforementioned automated installer file that is available from the **Releases** page.

If you wish to continue trying to install with vcpkg, the discussion in the link below may be useful for you if you have not seen it already.

https://github.com/IntelRealSense/librealsense/issues/6462

majinshaoyuindustry, 2021-01-04:
> Hi @majinshaoyuindustry You are correct that the front page of librealsense does document a vcpkg installation method:
> 
> https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg
> 
> In practice vcpkg is rarely used for librealsense installation though as far as I know, with the majority of Windows 10 users using the aforementioned automated installer file that is available from the **Releases** page.
> 
> If you wish to continue trying to install with vcpkg, the discussion in the link below may be useful for you if you have not seen it already.
> 
> [IntelRealSense/librealsense#6462](https://github.com/IntelRealSense/librealsense/issues/6462)

Hello MartyG, 
Thank you for this information, I finally managed to get it working using vcpkg on win10. It seems that there is no way to denote a specific version for vcpkg to install. I think the reason that running vcpkg install realsense2 from ROS will result to have the version 2.37 is that the vcpkg shipped with ROS is not the most up-to-date one. 
What is did is:
#1 remove vcpkg from the tools folder and git clone a fresh one. Then running vcpkg from ROS cmd resulted realsense2 2.40 which is very up-to-date (not 2.41 IDK why...) Then it was successfully complied with Realsense-ROS 2.2.20.

The fact that you can't use vcpkg like chocolatey or git to specify a version is very annoying in this situation...

But thanks for your help!

MartyG-RealSense, 2021-01-04:
You are very welcone - great news that you got it working.  Thanks for sharing your method!

MakerWangHQ, 2021-09-11:
> > Hi @majinshaoyuindustry You are correct that the front page of librealsense does document a vcpkg installation method:
> > https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg
> > In practice vcpkg is rarely used for librealsense installation though as far as I know, with the majority of Windows 10 users using the aforementioned automated installer file that is available from the **Releases** page.
> > If you wish to continue trying to install with vcpkg, the discussion in the link below may be useful for you if you have not seen it already.
> > [IntelRealSense/librealsense#6462](https://github.com/IntelRealSense/librealsense/issues/6462)
> 
> Hello MartyG,
> Thank you for this information, I finally managed to get it working using vcpkg on win10. It seems that there is no way to denote a specific version for vcpkg to install. I think the reason that running vcpkg install realsense2 from ROS will result to have the version 2.37 is that the vcpkg shipped with ROS is not the most up-to-date one.
> 
> What is did is:
> 
> Completely remove vcpkg from the tools folder and git clone a fresh one. Then running vcpkg from ROS cmd resulted realsense2 2.40 which is very up-to-date (not 2.41 IDK why...) Then it was successfully complied with Realsense-ROS 2.2.20.
> 
> The fact that you can't use vcpkg like chocolatey or git to specify a version is very annoying in this situation...
> 
> But thanks for your help!

 Do you install realsense2 using 'vcpkg' ?  Which version of realsense2 can be installed using vcpkg, is it detemined by ROS version?