realsenseai/realsense-ros #1510: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

kristfrizh, 2020-11-19:
Hii
I try to combine camera with lidar in one robot
I use laserscan_merger and depthimage_to_laserscan.
I keep getting the same error like this
 19/11 17:18:51,423 WARNING [139747262609152] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

I have see some error like this in issues but i have not find the solution.

this error also effect to my topic from laser realsense that sometimes appear and something not appear.
My assume that , this error occured because the data rate is faster than capacity data rate usb?
or anything else wrong?

My realsense is D435 and i use minicomputer linux 18.04

Thank you 
regards

MartyG-RealSense, 2020-11-19:
Hi @kristfrizh  If you are receiving the **control_transfer returned** warning only several times during launch and then they stop, or they occur one or two times a minute after launch, then this warning can be safely ignored.  You can check in **RViz** to see whether the topics are being published correctly despite the warnings.

If this warning is generating continuously though then it can indicate a serious communication problem with the device.

kristfrizh, 2020-11-19:
Hii MartyG
thank you for your reply
eventhough it just several times but it affect to my topic.
sometimes, the topic are not created and sometime it can be created.
Is it because the data rate too much to handle?

is there any way to fix it?
maybe provide a new usb hub?
or maybe can i use LAN to usb to provide a new hub?
(sory i am not quite expert with computer architecture, i am just guessing
what work and dont work)

MartyG-RealSense, 2020-11-19:
A good starting point in an investigation is to try adding a instruction to the end of your roslaunch statement that performs a reset of the 400 Series camera at launch.  The term to add is:

**initial_reset:=true**

MartyG-RealSense, 2020-11-26:
Hi @kristfrizh   Do you require further assistance with this case, please?   Thanks!

kristfrizh, 2020-11-26:
Hii Marty
sory for long updated

the behaviour still occurring.
I still not understand why the error happens
can you explain or give link?
what i thought the error happens because the data rate is way too high in
those bus,

do you have others suggestions?

should i buy a ethernet to USB module to create a new hub? (idk if it will
works or no)

thank you
really appreciate your response and help
Regards

MartyG-RealSense, 2020-11-26:
Hi @kristfrizh   Can you confirm if you have tried a hardware reset and still have the problem, please?

https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-730333695

kristfrizh, 2020-11-27:
Hii marty
I'm sorry but I cannot do that because I'm quite hard to access my usb port due to my mechanical enclosure.

MartyG-RealSense, 2020-11-27:
Hi @kristfrizh   I was not referring to resetting by physically unplugging the camera from the USB port, but instead adding a command to the roslaunch instruction that automatically resets the camera upon launch. 

https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-730333695

koikonomou, 2020-12-02:
Hi @MartyG-RealSense I have the same issue. I tried to add the command **initial_reset:=true**  to the end of my roslaunch statement as you suggested but nothing changed. Do you have any other suggestions ?

koikonomou, 2020-12-02:
Hi @MartyG-RealSense I have the same issue. I added the command initial_reset:=true to the end of my roslaunch statement as you suggested but nothing changed. Do you have any other suggestions ?

MartyG-RealSense, 2020-12-02:
Hi @koikonomou   Are you able to edit the FPS values inside the launch file to use 15 FPS instead of 30 to see if the reduction in FPS makes a difference to the number of **control_transfer returned** errors please?

kristfrizh, 2020-12-03:
hii marty, sory for the long update. 
I will try to do what you referring + the reduction FPS thing.
but right now, i quite bust with others thing
i will updated as soon as possible to you

thank you for your hospitality
regards

MartyG-RealSense, 2020-12-10:
Adding a note to keep this case open for a further time period.

Haitham96, 2020-12-17:
> hii marty, sory for the long update.
> I will try to do what you referring + the reduction FPS thing.
> but right now, i quite bust with others thing
> i will updated as soon as possible to you
> 
> thank you for your hospitality
> regards

Hi i have the same problem, how i can reduce the FBS?

MartyG-RealSense, 2020-12-17:
Hi @Haitham96   You can define a different FPS by editing the **arg** values in a roslaunch file.

https://stackoverflow.com/a/59726945

Haitham96, 2020-12-17:
> Hi @Haitham96 You can define a different FPS by editing the **arg** values in a roslaunch file.
> 
> https://stackoverflow.com/a/59726945

i tried to do reduce the fps to 15/6 , but i got the same error:

... logging to /home/haitham/.ros/log/17f6f94a-3fea-11eb-a4ad-40b0760db710/roslaunch-cggc-corsair270r-22894.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cggc-corsair270r:40355/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 71 lines cut ...]
[ INFO] [1608198914.141723800]: Done Setting Dynamic reconfig parameters.
[ INFO] [1608198914.142795189]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1608198914.144006474]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1608198914.144055173]: setupPublishers...
[ INFO] [1608198914.148379595]: Expected frequency for depth = 15.00000
[ INFO] [1608198914.195357460]: Expected frequency for color = 15.00000
[ INFO] [1608198914.218486064]: setupStreams...
[ INFO] [1608198914.240686780]: insert Depth to Stereo Module
[ INFO] [1608198914.240782898]: insert Color to RGB Camera
 17/12 11:55:14,306 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1608198914.357022708]: SELECTED BASE:Depth, 0
[ INFO] [1608198914.371193214]: RealSense Node Is Up!
 17/12 11:55:14,493 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,543 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,593 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,644 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,694 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 11:55:14,846 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,896 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,946 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 11:55:14,997 WARNING [140529039828736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

How i can fix it?

MartyG-RealSense, 2020-12-17:
If **control_transfer returned** is generating continuously even after launch then it can indicate that there is a serious communication problem with the device.  

I see that you have tried **initial_reset** already.  Does the launch improve if you add **enable_sync::=true** to your roslaunch instruction?

Haitham96, 2020-12-17:
No, i have the same errors:

... logging to /home/haitham/.ros/log/17f6f94a-3fea-11eb-a4ad-40b0760db710/roslaunch-cggc-corsair270r-25146.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cggc-corsair270r:38469/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
[... 77 lines cut ...]
[ INFO] [1608200036.153484160]: SELECTED BASE:Depth, 0
[ INFO] [1608200036.167799484]: RealSense Node Is Up!
 17/12 12:13:56,223 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,274 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,324 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,515 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,566 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,717 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,768 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,919 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:56,969 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:13:57,121 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 12:14:46,358 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:14:46,409 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:14:47,460 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:14:47,510 WARNING [139689348544256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Haitham96, 2020-12-17:
> If **control_transfer returned** is generating continuously even after launch then it can indicate that there is a serious communication problem with the device.
>
> I see that you have tried [...]

No, i have the same errors:

... logging to /home/haitham/.ros/log/17f6f94a-3fea-11eb-a4ad-40b0760db710/roslaunch-cggc-corsair270r-25900.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cggc-corsair270r:36635/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 68 lines cut ...]
[ INFO] [1608200165.543613112]: RGB Camera was found.
[ INFO] [1608200165.543714900]: num_filters: 0
[ INFO] [1608200165.543743352]: Setting Dynamic reconfig parameters.
[ INFO] [1608200165.972925669]: Done Setting Dynamic reconfig parameters.
[ INFO] [1608200165.974024805]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1608200165.975243254]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1608200165.975289168]: setupPublishers...
[ INFO] [1608200165.979183525]: Expected frequency for depth = 30.00000
[ INFO] [1608200166.028708638]: Expected frequency for color = 30.00000
[ INFO] [1608200166.062490042]: setupStreams...
[ INFO] [1608200166.084894858]: insert Depth to Stereo Module
[ INFO] [1608200166.084994270]: insert Color to RGB Camera
 17/12 12:16:06,150 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1608200166.200663543]: SELECTED BASE:Depth, 0
[ INFO] [1608200166.215346469]: RealSense Node Is Up!
 17/12 12:16:06,271 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:16:06,321 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:16:06,372 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:16:06,564 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 12:16:06,816 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:16:06,967 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:16:07,017 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 12:16:07,168 WARNING [140467358390016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2020-12-17:
Thank you very much for the information.  Are the topics published correctly in RViz despite the **control_transfer returned** warnings?

Haitham96, 2020-12-17:
> Thank you very much for the information. Are the topics published correctly in RViz despite the **control_transfer returned** warnings?

What you are mean by **topics published correctly in RViz** ?

MartyG-RealSense, 2020-12-17:
ROS has an interface called RViz that allows camera data to be displayed visually, a bit like the RealSense Viewer in librealsense.

If you are not familiar with using RViz in ROS, another way to check if the data is being published is to use a rostopic echo command:

**rostopic echo /camera/depth/image_rect_raw/header**

Haitham96, 2020-12-17:
> ROS has an interface called RViz that allows camera data to be displayed visually, a bit like the RealSense Viewer in librealsense.
>
> If you are not familiar with using RViz in ROS, another way to [...]

i tried to run this command **rostopic echo /camera/depth/image_rect_raw/header** and i got no output

[image]

MartyG-RealSense, 2020-12-17:
Can you provide the roslaunch instruction that you are using please.

Haitham96, 2020-12-17:
haitham@cggc-corsair270r:~$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_sync:=true
... logging to /home/haitham/.ros/log/17f6f94a-3fea-11eb-a4ad-40b0760db710/roslaunch-cggc-corsair270r-30689.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cggc-corsair270r:34263/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 405 lines cut ...]
 17/12 13:12:28,779 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:28,829 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:28,880 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:28,930 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,165 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,215 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,467 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,518 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,568 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,619 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 13:12:30,628 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:30,678 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:30,729 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:30,779 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Haitham96, 2020-12-17:
> Can you provide the roslaunch instruction that you are using please.

haitham@cggc-corsair270r:~$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_sync:=true
... logging to /home/haitham/.ros/log/17f6f94a-3fea-11eb-a4ad-40b0760db710/roslaunch-cggc-corsair270r-30689.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cggc-corsair270r:34263/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 407 lines cut ...]
 17/12 13:12:28,779 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:28,829 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:28,880 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:28,930 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,165 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,215 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,467 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,518 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,568 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:29,619 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 13:12:30,628 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:30,678 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:30,729 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:12:30,779 WARNING [140475587544832] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2020-12-17:
I meant the instruction that is used to launch, not the log.  Like this:

**roslaunch realsense2_camera rs_camera.launch**

Haitham96, 2020-12-17:
> I meant the instruction that is used to launch, not the log. Like this:
> 
> **roslaunch realsense2_camera rs_camera.launch**

in my previous answer i wrote each instruction with its output.

The instruction i used are:

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_sync:=true`

**with fps=15:**

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true depth_fps:=15 infra_fps:=15 color_fps:=15 enable_sync:=true`

**with fps=6**

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true depth_fps:=6 infra_fps:=6 color_fps:=6 enable_sync:=true`

MartyG-RealSense, 2020-12-17:
Okay, so you got no output when you checked if publishing was occurring using **rostopic echo /camera/depth/image_rect_raw/header**.   And I assume that you get no output either with a simple roslaunch instruction with no additional customization flags:

**roslaunch realsense2_camera rs_camera.launch**

Do you receive camera data normally in the RealSense Viewer program in librealsense?

Haitham96, 2020-12-17:
i didn't install RealSense Viewer program, just installed realsense-ros according to these instruction `https://github.com/IntelRealSense/realsense-ros`.
with the simple roslaunch instruction i got output, i got these log:

haitham@cggc-corsair270r:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/haitham/.ros/log/17f6f94a-3fea-11eb-a4ad-40b0760db710/roslaunch-cggc-corsair270r-32490.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cggc-corsair270r:42157/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 59 lines cut ...]
[ INFO] [1608205302.170507216]: num_filters: 0
[ INFO] [1608205302.170523790]: Setting Dynamic reconfig parameters.
[ INFO] [1608205302.575983334]: Done Setting Dynamic reconfig parameters.
[ INFO] [1608205302.576965206]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1608205302.578095928]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1608205302.578143207]: setupPublishers...
[ INFO] [1608205302.582137130]: Expected frequency for depth = 30.00000
[ INFO] [1608205302.632453779]: Expected frequency for color = 30.00000
[ INFO] [1608205302.660981375]: setupStreams...
[ INFO] [1608205302.683202958]: insert Depth to Stereo Module
[ INFO] [1608205302.683301382]: insert Color to RGB Camera
[ INFO] [1608205302.754466276]: SELECTED BASE:Depth, 0
[ INFO] [1608205302.767559287]: RealSense Node Is Up!
 17/12 13:41:42,895 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:41:42,945 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:41:43,096 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:41:43,163 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 13:41:43,516 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:41:43,566 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:42:03,534 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 13:42:03,585 WARNING [140243660990208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2020-12-17:
I think we have got to the point where, having tried everything else, the cause may be something that has gone wrong in the librealsense / ROS wrapper installation process.  There are cases where errors have just disappeared if a fresh installation of the operating system, librealsense and ROS wrapper is performed.

Haitham96, 2020-12-17:
> I think we have got to the point where, having tried everything else, the cause may be something that has gone wrong in the librealsense / ROS wrapper installation process. There are cases where [...]

What i should do?

MartyG-RealSense, 2020-12-17:
Let's go back to the installation documentation.  Did you use Method 1 or Method 2 to install the ROS wrapper please?

https://github.com/IntelRealSense/realsense-ros#installation-instructions

Haitham96, 2020-12-17:
> Let's go back to the installation documentation. Did you use Method 1 or Method 2 to install the ROS wrapper please?
> 
> https://github.com/IntelRealSense/realsense-ros#installation-instructions

i used Method 1: **Method 1: The ROS distribution**

MartyG-RealSense, 2020-12-17:
So all that you are doing in the installation is the two commands below to install librealsense and the ROS wrapper together?

export ROS_VER=melodic 
sudo apt-get install ros-$ROS_VER-realsense2-camera

Haitham96, 2020-12-17:
> So all that you are doing in the installation is the two commands below to install librealsense and the ROS wrapper together?
> 
> export ROS_VER=melodic
> sudo apt-get install ros-$ROS_VER-realsense2-camera

yes

MartyG-RealSense, 2020-12-17:
And what computer / computing device and operating system are you installing on?

For example, PC / Ubuntu?   Raspberry Pi / Raspbian OS?

Thanks!

Haitham96, 2020-12-17:
> And what computer / computing device and operating system are you installing on?
> 
> For example, PC / Ubuntu? Raspberry Pi / Raspbian OS?
> 
> Thanks!

Ubunto 18.04.5 LTS

MartyG-RealSense, 2020-12-17:
Ubuntu 18.04 LTS should be fine for installing from packages with Method 1.

How is the camera attached to the computer?  Plugged directly into a USB port on the PC or plugged into a USB hub?

Haitham96, 2020-12-17:
> Ubuntu 18.04 LTS should be fine for installing from packages with Method 1.
> 
> How is the camera attached to the computer? Plugged directly into a USB port on the PC or plugged into a USB hub?

Plugged directly into a USB port on the PC

MartyG-RealSense, 2020-12-17:
Do you have a different USB port you can try the camera in?  For example, if you are using a text-based ROS interface, unplug the mouse and plug the camera into that port instead.

Haitham96, 2020-12-17:
yes i have a different USB port, and tired to unplug the mouse and plug the camera into that port instead. and also changed the USB cable.
also i got the same error

MartyG-RealSense, 2020-12-17:
Can you input **realsense-viewer** into your Ubuntu terminal please to see whether the package installation of librealsense included the **RealSense Viewer** program?  If the Viewer does launch then you will be able to test whether the camera is able to stream.

Haitham96, 2020-12-17:
> Can you input **realsense-viewer** into your Ubuntu terminal please to see whether the package installation of librealsense included the **RealSense Viewer** program? If the Viewer does launch then you will be able to test whether the camera is able to stream.

it seems that the installation of librealsense doesn't included the RealSense Viewer program.
this is the output:

`haitham@cggc-corsair270r:~$ realsense-viewer
realsense-viewer: command not found`

MartyG-RealSense, 2020-12-17:
Okay.  Try **rs-enumerate-devices**

Haitham96, 2020-12-17:
> it seems that the installation of librealsense doesn't included the RealSense Viewer program.
> this is the output:
>
> [image]

also not found
[image]

MartyG-RealSense, 2020-12-17:
As far as I know, **rs-enumerate-devices** should be a standard part of a librealsense installation.   This suggests to me again that there is some problem with the Method 1 installation on your computer that is causing your difficulties.  I suspected this when you got no response at all from performing a rostopic echo, not even a message to say that there was no data from the topic.

https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-747370180

Haitham96, 2020-12-17:
> As far as I know, **rs-enumerate-devices** should be a standard part of a librealsense installation. This suggests to me again that there is some problem with the Method 1 installation on your computer that is causing your difficulties. I suspected this when you got no response at all from performing a rostopic echo, not even a message to say that there was no data from the topic.
> 
> [#1510 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-747370180)

what you suggest to do in this case?

MartyG-RealSense, 2020-12-17:
Move on to the **Method 2** installation, where librealsense 2.40.0 and wrapper 2.2.0 are built separately.  Librealsense first, and then the ROS wrapper secondly.

https://github.com/IntelRealSense/realsense-ros#method-2-the-realsense-distribution

Haitham96, 2020-12-17:
> Move on to the **Method 2** installation, where librealsense 2.40.0 and wrapper 2.2.0 are built separately. Librealsense first, and then the ROS wrapper secondly.
> 
> https://github.com/IntelRealSense/realsense-ros#method-2-the-realsense-distribution

How can i run the **.travis.yml** file?

MartyG-RealSense, 2020-12-17:
You can ignore the travis.yml file.  The installation instructions in Method 2 are going through the same steps that the travis.yml script describes.

Haitham96, 2020-12-17:
> You can ignore the travis.yml file. The installation instructions in Method 2 are going through the same steps that the travis.yml script describes.

I'm trying to follow the installation instruction here:
[image]

on the last command i run this command `modinfo uvcvideo | grep "version:"` and in my output there is no `realsense` string.
[image]

is it okay?

MartyG-RealSense, 2020-12-17:
This has happened to other people.

https://github.com/IntelRealSense/librealsense/issues/4815

I would think that if RealSense Viewer can be launched using **realsense-viewer** in **Ubuntu** then uvcvideo is likely okay.

Haitham96, 2020-12-17:
> This has happened to other people.
> 
> [IntelRealSense/librealsense#4815](https://github.com/IntelRealSense/librealsense/issues/4815)
> 
> I would think that if RealSense Viewer can be launched using the command **realsense-viewer** in Ubuntu and the Viewer can work successfully then uvcvideo is likely okay.

i installed through Method 2, after that i launched roslaunch using `roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_sync:=true` and i got the same error.
also i tried to run `rs-enumerate-devices` and i got an output

MartyG-RealSense, 2020-12-17:
What was the output from rs-enumerate-devices please?

Haitham96, 2020-12-17:
> What was the output from rs-enumerate-devices please?

haitham@cggc-corsair270r:~$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	727212071057
    Firmware Version              : 	05.12.09.00
    Recommended Firmware Version  : 	05.12.09.00
    Physical Port                 : 	2-6-56
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	828313021062
    Firmware Update Id            : 	828313021062

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 1	  848x100	@ 300Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 15Hz	   Y16
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
    Infrared 1	  640x360	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 90Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  424x240	@ 90Hz	   Y8
    Infrared 1	  424x240	@ 60Hz	   Y8
    Infrared 1	  424x240	@ 30Hz	   Y8
    Infrared 1	  424x240	@ 15Hz	   Y8
    Infrared 1	  424x240	@ 6Hz	   Y8
    Infrared 1	  256x144	@ 300Hz	   Y8
    Infrared 1	  256x144	@ 90Hz	   Y8
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 2	  848x480	@ 15Hz	   Y8
[... 231 lines cut ...]
    Color	  320x240	@ 30Hz	   YUYV
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV

MartyG-RealSense, 2020-12-17:
That looks okay.  Are you able to launch the RealSense Viewer program?

Haitham96, 2020-12-17:
> That looks okay. The camera is successfully detected and its details listed. Are you able to launch the RealSense Viewer program?

yes i can launch it using `realsense-viewer`

MartyG-RealSense, 2020-12-17:
Okay, that is a positive sign.  If the camera works in librealsense programs such as the Viewer then it does not necessarily mean that it will work in ROS, but at least it indicates that the camera hardware, the USB cable and librealsense are okay, which narrows it down to a ROS related probem.

Haitham96, 2020-12-17:
> Okay, that is a positive sign. If the camera works in librealsense programs such as the Viewer then it does not necessarily mean that it will work in ROS, but at least it indicates that the camera hardware, the USB cable and librealsense are okay, which narrows it down to a ROS related probem.

yes, what is should do?

[39 more replies]

closed 2020-12-31 by MartyG-RealSense