realsenseai/realsense-ros #1511: D435i No PointCloud2 and Map in rviz

ranjitkathiriya, 2020-11-19:
Hi Marty,

I have followed these steps everything working perfectly. Even I am able to see the video in rviz.
.
https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/

But I am not able to get D435i No PointCloud2 for and map also as I can see in this link :: https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

[image]
[image]

Thanks

MartyG-RealSense, 2020-11-19:
Hi @ranjitkathiriya   Since ROS wrapper 2.2.15 onward (which was published some time after the D435i SLAM guide), the IMU topics are disabled by default.  You can enable them in the roslaunch instruction by adding the commands below to the end of your roslaunch instruction:

enable_accel:=true enable_gyro:=true

Another example of setting up a point cloud in RViz is in the ROS wrapper documentation:

https://github.com/IntelRealSense/realsense-ros#point-cloud

ranjitkathiriya, 2020-11-20:
Hi Marty,

Thanks for the reply I have tried the following way then also it is not working for me. I will attach screenshot for firing the command, and add a map and my rviz environment. It will be helpful for you to troubleshoot the problem.

Thanks,
Ranjit
[image]
[image]
[image]

MartyG-RealSense, 2020-11-20:
What results do you get if you change Fixed Frame from 'map' to **camera_link** like the ROS wrapper's point cloud example does in its RViz image, please?

[image]

ranjitkathiriya, 2020-11-20:
Thank you so much again It is working doe me.

closed 2020-11-20 by ranjitkathiriya

MartyG-RealSense, 2020-11-20:
Great news - thanks!