realsenseai/realsense-ros #1512: Unable to publish Point Cloud from compressed Image

NRutigliano, 2020-11-19:
Hi everyone,
in the interest of a SLAM project realized in ROS, I am currently trying to develop a system that can merge depth data on a ground server. These data are produced by D435 cameras mounted on multiple robots. I would like to fuse them and produce a Point Cloud.
My first idea was to do that by sending a compressed image through a Wi-Fi connection. On the server, I decompress the image, convert it into a depth image and use it to produce a Point Cloud. By doing this for all the robots, I will have multiple point clouds to fuse to get a global one.
In order to do this, I was basically following the whitepaper at [**this link**](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-3-2-depth-image-recovery-from-colorized-depth-images-in-c).
Anyway, I didn't use the code provided in the paper references, but I started to write my own package.
So far, I managed to send the image topic /camera/depth/image_rect_raw/compressed in my network, I'm able to decompress the image and I developed a node to make the conversion to depth image and then to publish the point cloud. Anyway, if the decompression and the point cloud generation works like a charm, the conversion from colorized image to depth image keeps giving me problems and I cannot figure it out.

Therefore, here's my question(s):

- I would like to make this approach work, so is anyone able to tell me what I'm doing wrong?

- Has anyone already developed such a problem? How did you do it? Any suggestions, no matter what, could be very helpful.

Thank you!

Here's the code:
1. Python node to decompress the image: 
```

#!/usr/bin/env python2
#from _future_ import print_function
import sys, time
import cv2
import numpy as np
import roslib
import rospy
import message_filters
from cv_bridge import CvBridge, CvBridgeError
from sensor_msgs.msg import Image, CompressedImage, CameraInfo
from std_msgs.msg import String, Header
from std_msgs.msg import Float32MultiArray
from transforms3d.euler import euler2mat, mat2euler
from transforms3d.axangles import axangle2mat

class Decompressor:

	# Try to tune the queue_size value. Maybe increase it up to the frequency.	

	def __init__(self):
        	'''Initialize ros publisher, ros subscriber'''
        	# topic where we publish
        	self.image_pub  = rospy.Publisher("/output/image", Image,queue_size=1)

		self.bridge = CvBridge()
		
        	# subscribed Topic
        	self.subscriber = rospy.Subscriber("/camera/depth/image_rect_raw/compressed",
 CompressedImage,self.callback,queue_size=1)
		
	def callback(self,ros_data):

		image_np = self.bridge.compressed_imgmsg_to_cv2(ros_data,"bgr8")

        	out_img = self.bridge.cv2_to_imgmsg(image_np,"bgr8")
		out_img.header = ros_data.header
		#self.image_pub.publish(self.bridge.cv2_to_imgmsg(image_np,"rgb8"))
		self.image_pub.publish(out_img)		
		
class passthrough_class:
	def __init__(self):
		self.info_pub  = rospy.Publisher("/output/camera_info", CameraInfo,queue_size=1)
		self.info_sub = rospy.Subscriber("/camera/depth/camera_info", CameraInfo,self.callback,queue_size=1)
	def callback(self,data):
		self.info_pub.publish(data)

def main(args):
	'''Initializes and cleanup ros node'''
    	rospy.init_node('Decompressor', anonymous=True)
	ic = Decompressor()
	pt = passthrough_class()
[... 233 lines cut ...]
	 *iter_y = points[3*i+1];
	 *iter_z = points[3*i+2];
	 //std::cerr << *iter_x << " " << *iter_y << " " << *iter_z << std::endl;	// It display x,y,z points' coordinates on terminal.
	}

	cloud_pub.publish(cloud_msg);
	//std::cerr << ".";
    }    
	    catch (cv_bridge::Exception& e)
    {
      ROS_ERROR("cv_bridge exception: %s", e.what());
      return;
    }
  }
};

int main(int argc, char** argv)
{
  ros::init(argc, argv, "image_converter");
  ImageConverter ic;
  ros::spin();
  return 0;
}

```

MartyG-RealSense, 2020-11-19:
Hi @NRutigliano    I recall a past case where someone was having problems with extracting compressed images in ROS.  It was suggested to them that they may be publishing the decompressed images on the same topic as the originals.  So it may be worth checking that as a first step in this investigation.

https://answers.ros.org/question/298180/extracting-compressed-image-data-from-bag-file-results-in-doubled-up-or-dropped-frames/

******************

As a side-note: although your preference is a ROS-based solution, you may also be interested in a multi-camera ethernet network system developed by the CONIX Research Center at Carnegie Mellon that can stitch point clouds from up to 20 RealSense 400 Series cameras, each one on its own computer.

https://github.com/conix-center/pointcloud_stitching

NRutigliano, 2020-11-23:
Hi @MartyG-RealSense
Thanks for your fast reply, I checked the links. For what concerns the first one, I don't have a problem with the topic name, as you can see in the code that I posted early. Moreover, the issues with my package do not lie in extracting compressed images, but in converting a decompressed, colorized image into a depth grayscale image, as in the image below. 
[image]
The reason I opened the issue in the first place is that the colorized image on the camera side and the colorized image that I got once I decompressed the _**compressedImage**_ topic looks exactly the same, hence I figured out that the conversion should work just fine. But it doesn't.
I also started to check the second link, I might find something useful there. If I solve my issue, I will immediately write a comment here and let you know. In the meantime, if you have other suggestions, I would be very happy to hear it.
Thank you!

NRutigliano, 2020-11-23:
Ok thanks, I think I have enough material to investigate. I will try not to use the compressed topic. Still, I don't get why the colorization method doesn't work. Anyway, I'll let you know how it goes. Thanks, again.

NRutigliano, 2020-11-27:
UPDATE:
I managed to improve the quality of the reconstructed depth image, therefore the Point cloud. To achieve this I double-checked all the post-processing filters used in the camera and by reconfiguring the parameter **_histogram equalization_** to false in the colorizer filter, the reconstructed depth image has gotten way better. Anyway, I think it's possible to do better, so I will keep trying. I will update as soon as I get new results

MartyG-RealSense, 2020-11-27:
Great to hear that you achieved improvement.  Thanks for sharing details of the method that worked for you, and please do update further after your new tests.   Good luck!

MartyG-RealSense, 2020-12-04:
Hi @NRutigliano   Do you require further assistance with this case, please?   Thanks!

NRutigliano, 2020-12-04:
Hi @MartyG-RealSense, I can take care of it by myself now. Thanks!

closed 2020-12-04 by NRutigliano

MartyG-RealSense, 2020-12-04:
Great to hear @NRutigliano  - thanks very much for the update!