realsenseai/realsense-ros #1513: D435i process die after update

dtquoc97, 2020-11-19:
Hi,
I have just updated realsense SDK v2.40.0 and then I reinstall realsense-ros.
When I run roslaunch realsense2_camera rs_camera.launch, I receive 

[image]

Thanks

MartyG-RealSense, 2020-11-19:
Hi @dtquoc97   Did you build from packages using **Method 1** of the ROS wrapper installation instructions, please?

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

 If so, I would recommend using Method 2 (building the ROS wrapper from source code) for the moment in case the packages are not yet updated to SDK 2.40.0, as this SDK version was only released yesterday.  If SDK 2.40.0 is already installed, skip to Step 2 of Method 2 to install the RealSense ROS wrapper form source:

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

dtquoc97, 2020-11-19:
After update SDK, I try to run Slam with D435i but it don't receive point cloud. So reinstall ROS wrapper from source and get this problem,
There is a issue  i have two rule in etc/udev/ and lib/udev, so i remove one in etc/edev. Is it have any problem

MartyG-RealSense, 2020-11-19:
The **etc** rule is the one that can be removed safely to solve the **multiple udev rules** warning.  The **lib** rule should not be removed, as doing so can break the linkages of Debian packages.

dtquoc97, 2020-11-19:
I just install ROS wrapper in another workspace but it still not working

MartyG-RealSense, 2020-11-20:
What errors do you receive now please?

dtquoc97, 2020-11-20:
Although I used method 2 to build, this problem has not been resolved. 
I try to run with older version of SDK and ROS wrapper on another PC, it have the same problem.

I think it may be relate to camera's firmware. 
After update SDK and connect with camera, it ask me update firmware and i saw this error after run enable_accel:=true enable_gyro:=true  to resolved issue https://github.com/IntelRealSense/realsense-ros/issues/1511

MartyG-RealSense, 2020-11-20:
The problem in that link was solved:

https://github.com/IntelRealSense/realsense-ros/issues/1511#issuecomment-731260035

dtquoc97, 2020-11-24:
The problem was not caused by realsense-ros. I uninstall realsense SDK and install it again instead update.

Thank for your support

closed 2020-11-24 by dtquoc97