realsenseai/realsense-ros #1514: How to get RGB Depth stream in python ROS

ranjitkathiriya, 2020-11-19:
Hi @doronhi,

Actually, I am getting RGB-D from the stream but it is in black color can you please help me with that here I am posting my code and Image of the output.

```
#!/usr/bin/env python3
import rospy
from sensor_msgs.msg import Image
from cv_bridge import CvBridge 
import cv2
import numpy as np
 
def callback(data):
  br = CvBridge()
  rospy.loginfo("receiving video frame")
  current_frame = br.imgmsg_to_cv2(data)
  cv_image_array = np.array(current_frame, dtype = np.dtype('f8'))
  cv_image_norm = cv2.normalize(cv_image_array, cv_image_array, 0, 1, cv2.NORM_MINMAX)
  cv2.imshow("camera", cv_image_norm)
  print(cv_image_norm.shape)
  cv2.waitKey(1)
      
def receive_message():
  rospy.init_node('video_sub_py', anonymous=True)
  rospy.Subscriber('/camera/aligned_depth_to_color/image_raw', Image, callback)
  rospy.spin()
  cv2.destroyAllWindows()
  
if __name__ == '__main__':
  receive_message()
```

[image]

Thank you so much for the help

MartyG-RealSense, 2020-11-19:
Hi @ranjitkathiriya   I wonder if you need to define a Colorizer filter in your ROS launch.

https://github.com/IntelRealSense/realsense-ros/issues/501

ranjitkathiriya, 2020-11-20:
Thanks, Solved::

by running this ros command:
roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true filters:=colorizer align_depth:=true

closed 2020-11-20 by ranjitkathiriya

MartyG-RealSense, 2020-11-20:
Great news, thanks for the update!