realsenseai/realsense-ros #1515: Multiple Camera

BunnyIcon, 2020-11-20:
I'm a beginner with computer vision and embedded control. I'm doing a robot project that needs 6 cameras (six D435i or one D435i and five T265). I'm using Jetson AGX Xavier kits. It provides 3 usb3.0. I want to know how to increase the USB ports so that I could use more cameras. I tried with miniPCIE, and USB HUB, but neither works. 

Do I need a powered USB hub? And how many cameras can be processed at most by Jetson AGX Xavier?

Thanks in advance!

RealSenseSupport, 2020-11-20:
@BunnyIcon, please give multi-camera launch file a try. Make sure you supply the correct camera SN accordingly.
https://github.com/IntelRealSense/realsense-ros#work-with-multiple-camera
Other users also have reported successful experience of running multiple RealSense cameras with PCIe card on AGX Xavier.
Ref: https://github.com/IntelRealSense/realsense-ros/issues/1018#issuecomment-584950118

BunnyIcon, 2020-11-28:
> @BunnyIcon, please give multi-camera launch file a try. Make sure you supply the correct camera SN accordingly.
> https://github.com/IntelRealSense/realsense-ros#work-with-multiple-camera
> Other users also have reported successful experience of running multiple RealSense cameras with PCIe card on AGX Xavier.
> Ref: [#1018 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1018#issuecomment-584950118)

Hello, the situation even got worse. I tried to insert 3 cameras on Jetson AGX Xavier directly, but failed.
Here are my multi-camera launch file and the report in terminals.

<launch>
  <arg name="serial_no_camera1"    			default="036222071741"/> 			
  <arg name="serial_no_camera2"    			default="036522071177"/> 			
  <arg name="serial_no_camera3"    			default="036422070801"/> 			
  <arg name="camera1"              			default="camera1"/>		
  <arg name="camera2"              			default="camera2"/>		
  <arg name="camera3"              			default="camera3"/>		
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  <arg name="tf_prefix_camera3"         default="$(arg camera3)"/>
  <arg name="initial_reset"             default="false"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <group ns="$(arg camera2)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera2)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera2)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <group ns="$(arg camera3)" if="$(eval serial_no_camera3 != '')">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera3)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera3)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>
</launch>

nvidia@miivii-tegra:~$ roslaunch realsense2_camera rs_multiple_devices.launch 
... logging to /home/nvidia/.ros/log/ed45c9d0-314b-11eb-936a-48b02d2b89d0/roslaunch-miivii-tegra-17086.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://miivii-tegra:45045/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
[... 1886 lines cut ...]
[ 2746.078883] usb 2-3: SerialNumber: 035323051017
[ 2746.081450] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2746.081467] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2746.081721] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 2746.087273] uvcvideo: Unable to create debugfs 2-9 directory.
[ 2746.087465] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 2746.087858] uvcvideo 2-3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2746.088160] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 2746.088600] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2746.089115] input: Intel(R) RealSense(TM) Depth Ca as /devices/3610000.xhci/usb2/2-3/2-3:1.0/input/input21
[ 2746.090348] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 2746.090365] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 2746.093059] uvcvideo: Unable to create debugfs 2-9 directory.
[ 2746.093248] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!
[ 2746.093554] uvcvideo 2-3:1.3: Entity type for entity Extension 8 was not initialized!
[ 2746.093856] uvcvideo 2-3:1.3: Entity type for entity Camera 6 was not initialized!
[ 2752.266116] usb 2-4: usb_suspend_both: status 0
[ 2752.266126] usb 2-1: usb_suspend_both: status 0
[ 2752.266131] usb 2-3: usb_suspend_both: status 0
[ 2752.266269] usb usb2: usb_suspend_both: status 0
[ 2772.426120] usb 2-4: usb_suspend_both: status 0
[ 2772.426140] usb 2-1: usb_suspend_both: status 0
[ 2772.430132] usb 2-3: usb_suspend_both: status 0
[ 2772.430282] usb usb2: usb_suspend_both: status 0
[ 2797.276481] nvgpu: 17000000.gv11b             railgate_enable_store:283  [INFO]  railgate is disabled.

BunnyIcon, 2020-11-29:
@RealSenseSupport

BunnyIcon, 2020-11-30:
@RealSenseSupport 

Thanks for your help. I tried multi device launch file, and got one more problem: when I used 6 D435i simultaneously, it often turns out some (one or two or three) cameras cannot publish /color/ topics. But after a period of time, when used alone, each camera can publish all topics. Sometimes this problem occurs, sometimes not. It seems quite unstable.

I'm using 2 USB hubs with power (3 cameras on one hub): https://detail.tmall.com/item.htm?id=599038479527&spm=a1z09.2.0.0.47902e8dApIilm&_u=u2a12faqaa76&sku_properties=148242406:21516
(The link may be in Chinese, sry for hard to read)

Do you have any ideas for the unstable reason? Thanks in advance.

RealSenseSupport, 2020-12-03:
@BunnyIcon, was the librealsense built using RSUSB backend or the native [v4l backend](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend)?
For multi camera configuration, please build the library using the native backend. 
See the RSUSB limitation [here](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400).

BunnyIcon, 2020-12-03:
@RealSenseSupport That might be the point! How could I check whether I built using RSUSB backend?

darshb34, 2020-12-09:
@RealSenseSupport 

1.) I have same issue on same platform, I installed via sudo apt-get install ros-melodic-realsense2-camera (it automatically installs librealsense). Does that install as native v4l backend?

2.) I have two robots with same setup and same installation process for ros-melodic-realsense2-camera, but in one of them I have v4l paths for rs cams and not in the other one.

Please share your thoughts

RealSenseSupport, 2020-12-10:
Please check the message generated by the RealSense Viewer log which shows the backend information.
[image]
For information on how to build the libRealSense with native backend, please use v2.40.0 and the instruction provided here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

doronhi, 2020-12-14:
As for now, installation using `apt-get install ros-melodic-realsense2-camera` install a version of librealsense2 that uses RSUSB, not v4l.

RealSenseSupport, 2020-12-22:
Please let us know if more help is needed. Thank you.

closed 2020-12-28 by RealSenseSupport