realsenseai/realsense-ros #1516: Only shows depth topics.

konnoeric, 2020-11-20:
I launched "roslaunch realsense2_camera rs_camera.launch" and it shows only depth topics. How can I see RGB and Lidar pointcloud topics?
I am using Intel RealSense L515. Ubuntu 18.04.5 LTS.  Intel® Core™ i5-4590 CPU.
[image]

doronhi, 2020-11-22:
It could help if you published the log of the realsense2_camera node. It looks like using parameter enable_color:=false. To publish a pointcloud you'll have to set enable_pointcloud:=true either in the command line on in the rs_camera.launch file.

RealSenseSupport, 2021-01-12:
Hi @konnoeric 

Is there anything else you need on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-19 by RealSenseSupport