realsenseai/realsense-ros #1519: How to get a frame_id during ros subscribe the /camera/color/image_raw channels

ranjitkathiriya, 2020-11-24:
Hi there,

I need to get the frame id of ros after subscribing to the /camera/color/image_raw  - channels.

I have tried this command from your document:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true filters:=colorizer enable_pointcloud:=true base_frame_id:=true`

But how can I see in the terminal - in other words, how can I listen to it that which frame is currently running.

Thanks

MartyG-RealSense, 2020-11-24:
Hi @ranjitkathiriya   The ROS discussion linked to below suggests using **rostopic echo** to listen to a particular topic, and add the term **|grep frame_id** to only display the frame_id details.

https://answers.ros.org/question/42632/how-to-know-the-frame-id/?answer=42633#post-id-42633

ranjitkathiriya, 2020-11-24:
Thanks Agan

closed 2020-11-24 by ranjitkathiriya