realsenseai/realsense-ros #1520: ROS wrapper melodic windows  build package issue

esteban-andrade, 2020-11-24:
Hi all, 
I have installed the Realsense SDK on windows succesfully. However, when i try to build the ROS wrapper package for ros melodic on windows 10. I am getting the following error message. 

I have installed the SDK using the installed from Github. Is there any solution out there that could allow me to address this issue?

```
C:\Users\esgaa\Desktop\ROS\catkin_workspace>catkin_make clean
Base path: C:\Users\esgaa\Desktop\ROS\catkin_workspace
Source space: C:\Users\esgaa\Desktop\ROS\catkin_workspace\src
Build space: C:\Users\esgaa\Desktop\ROS\catkin_workspace\build
Devel space: C:\Users\esgaa\Desktop\ROS\catkin_workspace\devel
Install space: C:\Users\esgaa\Desktop\ROS\catkin_workspace\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\Users\esgaa\Desktop\ROS\catkin_workspace\build"
####

Microsoft (R) Program Maintenance Utility Version 14.28.29333.0
Copyright (C) Microsoft Corporation.  All rights reserved.

-- Using CATKIN_DEVEL_PREFIX: C:/Users/esgaa/Desktop/ROS/catkin_workspace/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\melodic\x64\tools\vcpkg\installed\x64-windows;C:/opt/ros/melodic/x64
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/ros/melodic/x64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/ros/melodic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:41 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/Users/esgaa/Desktop/ROS/catkin_workspace/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/esgaa/Desktop/ROS/catkin_workspace/build/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'C:\opt\ros\melodic\x64\lib\site-packages\cmake\data\bin\cmake.exe' : return code '0x1'
Stop.
Invoking "nmake cmake_check_build_system" failed

```

Many Thanks

esteban-andrade, 2020-11-24:
hi @MartyG-RealSense  That is a simbolink link if i decide to build in and do catkin_make in the same directory the same error appears. By same direcectory is navigating to the catkin_ws and build it there

```
[C:\catkin_ws>catkin_make clean
Base path: C:\catkin_ws
Source space: C:\catkin_ws\src
Build space: C:\catkin_ws\build
Devel space: C:\catkin_ws\devel
Install space: C:\catkin_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\catkin_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.28.29333.0
Copyright (C) Microsoft Corporation.  All rights reserved.

-- Using CATKIN_DEVEL_PREFIX: C:/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\melodic\x64\tools\vcpkg\installed\x64-windows;C:/catkin_ws/devel;C:/opt/ros/melodic/x64
-- This workspace overlays: C:/catkin_ws/devel;C:/opt/ros/melodic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/ros/melodic/x64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/ros/melodic/x64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/melodic/x64/Scripts/nosetests-2.7.exe
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:41 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/catkin_ws/build/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'C:\opt\ros\melodic\x64\lib\site-packages\cmake\data\bin\cmake.exe' : return code '0x1'
Stop.
Invoking "nmake cmake_check_build_system" failed

C:\catkin_ws>
Stop.
Invoking "nmake cmake_check_build_system" failed
```

MartyG-RealSense, 2020-11-24:
Hi again @esteban-andrade   Apologies for the delay in responding further, as I was researching your case further.

Sometimes the clean proces has caused problems for people.  A RealSense ROS user posted in the link below a version of the wrapper setup process that worked for them.

https://github.com/IntelRealSense/realsense-ros/issues/1432#issuecomment-706195151

I will also bring the RealSense ROs wrapper developer into this case for his insights.

@doronhi  Could you kindly provide your thoughts on this case please?   Thanks!

esteban-andrade, 2020-12-01:
Hi @MartyG-RealSense , 

Yes I added the ddynamic_reconfigure package. However it still had the same issue. 
The only way that it worked was by using the above command

`rosdep install --from-paths src --ignore-src -r -y`

MartyG-RealSense, 2020-12-01:
Thanks very much @esteban-andrade for the update.  Is RealSense ROS usable for you if you use the **rosdep** method above, please?

esteban-andrade, 2020-12-01:
Hi, @MartyG-RealSense  yes with that command the launch files are able to work. However, I found that most of the time before I start, I have to catkin_make again even after doing devel\setup.bat because it can't recognize it.

MartyG-RealSense, 2020-12-02:
I conducted further research into your case.  I could not find many who had this problem.   Of the ones that did, most of them resorted to building the ROS wrapper from source code to get it working.

The rest of them followed the advice in the case linked to below:

https://github.com/IntelRealSense/realsense-ros/issues/280

closed 2020-12-09 by MartyG-RealSense