realsenseai/realsense-ros #1521: dkms won't install on kernel 5.4

mcamurri, 2020-11-24:
Hi all, 
I'm following the instructions from here 
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

to install the newest release of the librealsense. The package version is correcty 2.40

However, when it's installing the dkms module I receive the following error:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libnvidia-common-450
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed
  librealsense2-dkms librealsense2-udev-rules
0 to upgrade, 2 to newly install, 0 to remove and 0 not to upgrade.
Need to get 0 B/10.2 MB of archives.
After this operation, 54.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 466007 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.14-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.14-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.40.0-0~realsense0.3814_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.40.0-0~realsense0.3814) ...
Setting up librealsense2-udev-rules:amd64 (2.40.0-0~realsense0.3814) ...
Setting up librealsense2-dkms (1.3.14-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.14 DKMS files...
Building for 5.4.0-54-generic
Building initial module for 5.4.0-54-generic
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
```

By analyzing the content of the mentioned `dkms.conf` from the `.deb` file, it seems only kernels up to 5.3 are supported:
```
grep -r BUILD_EXCLUSIVE .
./data/usr/src/librealsense2-dkms-1.3.14/dkms.conf:BUILD_EXCLUSIVE_KERNEL="^(4.1([5|8]).0-*|5.[0|3])"
```
however, the  documentation says that the kernel 5.4 is supported:
https://github.com/IntelRealSense/librealsense/releases

I'll stick with the latest 5.3 for now.

MartyG-RealSense, 2020-11-24:
Hi @mcamurri   Although official kernel 5.4 support was added in SDK 2.40.0, the **5.4-generic** kernel that you are using has worked successfully with librealsense for some months and had stable performance.  

Would building librealsense from source code instead of packages be an option for you to try, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

mcamurri, 2020-11-24:
Thank you @MartyG-RealSense , at the moment I don't have a strict need for the 5.4, so I will stick to the 5.3 and apt package. 
I just noticed that it didn't work with the latest kernel, so I thought it might of interest for you to know.
If the 5.4 works then I guess it's just a matter of adding a `4` instead of a `3` in that `dkms.conf` file.

MartyG-RealSense, 2020-11-24:
I will pass your experience on to Intel.   Thanks for the report!   If you are satisifed with the outcome, please feel free to close the case with the **Close with comment** button under the comment writing box.

closed 2020-11-26 by mcamurri

HartmannSa, 2020-12-23:
Hi all,

I have the same Error as @mcamurri when installing librealsense2-dkms.
My system is Ubuntu 18.04 LTS.
My Kernel is 5.4.0-58-generic.

@mcamurri , you mentioned that changing the 3 to a 4 in dkms.conf could solve the issue.
How can I do this? Where do I find "BUILD_EXCLUSIVE" and dkms.conf?

Changing this Numbers seems easier to me than downgrading the kernel.

I look forward to an answer.

HartmannSa, 2020-12-23:
Hi again,

I solved some of my questions on my own with the help of IntelRealSense/librealsense#7995:
`cd /usr/src/librealsense2-dkms-1.3.14` and `grep -r BUILD_EXCLUSIVE .`

But I guess that the number change can only be done by the developers?