realsenseai/realsense-ros #1522: Path Planning

intel-user, 2020-11-25:
Hi, I'm trying to perform global path plan while I have goal position on ROS with move_base package, something like this:

[image]

But I don't know how to create move_base node with realsense-ros package, Can you give me some instruction for that?
Thank you!!

MartyG-RealSense, 2020-11-25:
Hi @intel-user   My research found a paper in PDF format that uses a RealSense D435 camera with the RealSense ROS wrapper and move_path.

https://www.utupub.fi/bitstream/handle/10024/147661/VSLAM_and_Navigation_System_of_Unmanned_Ground_Vehicle_Based_on_RGB-D_Camera.pd

It also discusses previous generations of RealSense camera model such as the R200.   So for ease of reading, the D435 information starts on page 29 and the move_base inavigation information on page 35 onwards.

If you are able to consider an alternative to using move_base for navigation, the **Erwhi Hedgehog** project provides an excellent and user-friendly blueprint for a ROS mobile robot equipped with a 400 Series camera.

https://www.intelrealsense.com/hedgehog-robot-by-erwhi/

https://gbr1.github.io/erwhi_hedgehog.html

If you need to use move_base in your project, the **SAWR** project provides another complete RealSense robot blueprint that provides details of setting up move_base.  The project is somewhat old and pre-dates the 400 Series cameras, though the author believed that it would be straightforward to update the project for those camera models.

https://software.intel.com/content/www/us/en/develop/articles/build-an-autonomous-mobile-robot-with-the-intel-realsense-camera-ros-and-sawr.html

https://github.com/intel/sawr/blob/master/sawr_master/INSTALL.md

intel-user, 2020-11-27:
Thank you so much

MartyG-RealSense, 2020-12-03:
Hi @intel-user   Do you require further assistance with this case, please?   Thanks!

intel-user, 2020-12-07:
I install move_base ros package and download the params folder on the internet and then,  I add move_base node into opensource_tracking.launch file. It's work for me.

Thank for your help!!

MartyG-RealSense, 2020-12-07:
That's great to hear @intel-user  - thanks so much for the update!

As you had a successful outcome, I will close this case.  Thanks again!

closed 2020-12-07 by MartyG-RealSense