realsenseai/realsense-ros #1523: How to get RGB image from realsense L515 depth camera

kashishdhal, 2020-11-25:
Hello,

I am using the L515 camera to get the point cloud with the launch file **rs_camera.launch**. However, I want to get an RGB image along with the depth image. But I am not able to see **/camera/color/image_raw** topic being published. I am only able to see the depth data coming over the ROS topics.

Please help me to get the RGB image.

Regards,
Kashish Dhal

yuzhou42, 2020-11-28:
Hi Kashish, I just figure this out. In order to get the l515 color image, you will need to change the image size in the launch file. Because l515 doesn't support 640x480. You can choose anyone from 960x540, 1280x720 and 1920x1080.

Yu

kashishdhal, 2020-11-28:
Hello @yuzhou42 ,

Thanks for the reply! One question, will that also change the resolution of the depth image?

yuzhou42, 2020-11-28:
Hi @kashishdhal, it won't change the resolution of depth. They are using different parameters.

kashishdhal, 2020-12-01:
Hello @yuzhou42 ,

Thanks, it worked!

closed 2020-12-01 by kashishdhal

karanpandyaa, 2024-05-19:
I had same issue but the rgb topics got published when i ran:
 **roslaunch realsense2_camera rs_aligned_depth.launch**