realsenseai/realsense-ros #1524: /camera/depth/image_rect_raw to depth values

miajl, 2020-11-25:
Hello, I am having trouble figuring out how to convert the images on /camera/depth/image_rect_raw to actual depth values. I know that for a 16bit depth, one unit is 1 mm, but ros image data is made of uint8. How does that change the conversion?

MartyG-RealSense, 2020-11-26:
Hi @miajl   One approach that a RealSense user applied successfully was to convert the **/camera/depth/image_rect_raw** image to an OpenCV image using **cv_bridge** and then use obtain the depth value using a **double distance** calculation.

https://answers.ros.org/question/299888/using-a-depth-map/

yuzhou42, 2020-11-28:
Hi @miajl, the following method works for me.

```Python
import rospy
from cv_bridge import CvBridge, CvBridgeError
import sensor_msgs.msg  
import numpy as np
from PIL import Image
i = 0
root='/home/yu'
def convert_depth_image(ros_image):
    bridge = CvBridge()
    global i
    depth_image = bridge.imgmsg_to_cv2(ros_image, desired_encoding="passthrough")
    depth_array = np.array(depth_image, dtype=np.float32)
    im = Image.fromarray(depth_array)
    im = im.convert("L")
    idx = str(i).zfill(4)
    im.save(root+"/depth/frame{index}.png".format(index = idx))
    i += 1
    print("depth_idx: ", i)

def pixel2depth():
    rospy.init_node('pixel2depth',anonymous=True)
    rospy.Subscriber("/camera/aligned_depth_to_color/image_raw", sensor_msgs.msg.Image,callback=convert_depth_image, queue_size=10)
    rospy.spin()

if __name__ == '__main__':
    pixel2depth()
```

Hope it helps!

Yu

MartyG-RealSense, 2020-11-28:
Hi @yuzhou42   Thanks so much for your help with the question of @miajl and contributing your script for RealSense community members to benefit from in future    :)

yuzhou42, 2020-11-28:
@MartyG-RealSense  You're welcome! Glad it helps  ^_^

doronhi, 2020-11-29:
Thanks @yuzhou42 . For future reference, users can also check out the following python script included in this package: [show_center_depth.py](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py). It demonstrates subscribing to depth image and depth camera info topics and extracting depth and 3D position in camera's coordinates.

miajl, 2020-11-30:
Would doing this get the correct depth values? It looks right when I plot it but I want to make sure.

```
bridge = CvBridge()
cv_img = bridge.imgmsg_to_cv2(ros_img, "passthrough")
depth_array = np.array(cv_img, dtype=np.uint16)*0.001
```

MartyG-RealSense, 2020-12-07:
Hi @doronhi   Can you take a look at the question above from @miajl please?   Thanks!

MartyG-RealSense, 2020-12-14:
Hi @miajl    Did you find a solution to your question above, please?

MartyG-RealSense, 2020-12-21:
Case closed due to no further comments received.

closed 2020-12-21 by MartyG-RealSense

deadln, 2022-04-08:
> Would doing this get the correct depth values? It looks right when I plot it but I want to make sure.
> 
> ```
> 
> ```

Better use 0.1 multiplier