realsenseai/realsense-ros #1525: Getting Error while running python code with real-sense 435i

ranjitkathiriya, 2020-11-26:
Hi there,

I am getting this error every thinks in the code is working fine but I don't know that this error can cause harm to my code or not please! give a clarification on this error. 

```
26/11 11:21:50,540 WARNING [139840600119040] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
 26/11 11:21:50,602 ERROR [139840600119040] (backend-v4l2.h:194) xioctl(VIDIOC_QBUF) guard failed for fd 18
 26/11 11:21:50,602 WARNING [139840600119040] (backend-v4l2.h:197) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 18
 26/11 11:21:50,602 WARNING [139840600119040] (backend-v4l2.h:201) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 18
 26/11 11:21:50,602 ERROR [139840600119040] (backend-v4l2.h:194) xioctl(VIDIOC_QBUF) guard failed for fd 18
 26/11 11:21:50,602 WARNING [139840600119040] (backend-v4l2.h:197) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 18
 26/11 11:21:50,602 WARNING [139840600119040] (backend-v4l2.h:201) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 18
 26/11 11:21:50,602 ERROR [139840600119040] (backend-v4l2.h:194) xioctl(VIDIOC_QBUF) guard failed for fd 18
 26/11 11:21:50,603 WARNING [139840600119040] (backend-v4l2.h:197) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 18
 26/11 11:21:50,603 WARNING [139840600119040] (backend-v4l2.h:201) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 18
 26/11 11:21:50,603 ERROR [139840600119040] (backend-v4l2.h:194) xioctl(VIDIOC_QBUF) guard failed for fd 18
 26/11 11:21:50,603 WARNING [139840600119040] (backend-v4l2.h:204) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 18
 26/11 11:21:50,723 ERROR [139840600119040] (backend-v4l2.cpp:412) Non-sequential Video and Metadata v4l buffers
 26/11 11:21:50,723 WARNING [139840600119040] (backend-v4l2.cpp:1019) Video frame dropped, video and metadata buffers inconsistency
 26/11 11:21:50,783 ERROR [139840600119040] (backend-v4l2.cpp:412) Non-sequential Video and Metadata v4l buffers
 26/11 11:21:50,783 WARNING [139840600119040] (backend-v4l2.cpp:1019) Video frame dropped, video and metadata buffers inconsistency
```

Thanks.

MartyG-RealSense, 2020-11-26:
Hi @ranjitkathiriya   I was able to find one previous case of these errors.   The solution reported by a RealSense user was to update all of the librealsense Debian packages.

https://github.com/IntelRealSense/realsense-ros/issues/1142

closed 2020-12-02 by ranjitkathiriya

ranjitkathiriya, 2020-12-03:
Thanks for help

MartyG-RealSense, 2020-12-03:
You are very welcome @ranjitkathiriya  - thanks for the update!