realsenseai/realsense-ros #1526: Odometry reference with T265

pfedom, 2020-11-27:
Hello, 
I am using the Realsense T265 as an odometry source of my Robot. (Ros2 Eloquent)
First of all, thank you for the eloquent branch!

The camera is mounted on the front of the Robot (40cm in front of the center). Therefore I have a setup question. 
If I turn in one spot, I want the odometry output to sit in the base of the robot and also rotate around itself. 
If I do so at the moment, the camera shows its own movement, not referenced to the robot base. The odometry output creates a circle with the radius of about 40 (the radius of the robot).

The odometry message has header frame: `odom` and child frame: `base_link`. 
I tried to set a static_tf from `base_link` to  `camera_pose_frame` to let the system know where the camera is mounted. 
Unfortunately this didnt solve my issue. 

Is it possible to get it working correct with TFs or do I have to go with the odometry_input config to tell the camera where it is mounted on the robot to get the result of a correct odometry referenced on the robots base?

Thanks in advance!

doronhi, 2020-11-29:
I'm not sure how come the odom message has a child frame: `base_link`. Did you rename?
The output of `/camera1/odom/sample` shows frame_id: odom_frame, child_frame_id: camera_pose_frame.
`odom_frame`, can be renamed using parameter `odom_frame_id` refers to the origin point of the camera. The position values in the message refer to that point.
Setting a static_tf wont change the values in the message.
What you should do, I think, is define a new frame for the center of the robot and set a static_tf so that `robot_center_frame` will depend on `camera_pose_frame` (The static_tf is something with translation of 0.4,0,0 and angles of 0,0,0 but I could be wrong here).
I didn't try it but I think that would be a more promising direction then the config file.

pfedom, 2020-11-29:
Yes I renamed it in the process. Totally forgot about that. 
I will try yours out and let you know. 
The T265 odometry is my only odometry source at the moment but with static tf i should be able to build the tree how I want. 

Maybe its possible to add a new base frame in the constants.h where you can tell the realsense-node what the base of the robot is. This would solve the issue wouldn't it?

pfedom, 2020-11-30:
Update: 
A static tf from `camera_pose_frame` to `robot_base` helps. The robot base frame moves correctly and as expected. 
The odometry message of course still references into `camera_pose_frame` but I dont think thats an issue. 
To fix this I would have to create an own new tf publisher to handle it. 

Thanks for your help @doronhi

closed 2020-11-30 by pfedom