realsenseai/realsense-ros #1528: Reflex object issue

impet14, 2020-11-27:
Hi,
Is there any idea to resolve the reflex object issue?
Since I cannot get point cloud on the reflex area. 
Thank you

MartyG-RealSense, 2020-11-27:
Hi @impet14   Do you mean that you are having problems getting a point cloud in areas where there is **reflections** please?

If this is correct then **Section 4.4** of Intel's paper on physical optical filters that can be applied over the top of camera lenses on the outside of the camera has advice about this.   It recommends using a **libear polarizer** lens.  This can greatly reduce glare from reflections.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

[image]

If it is a small object that the camera is observing then another way to dampen reflections is to apply a fine spray-on powder such as foot powder or baby powder onto the surface of the object.   You can also purchase a professional 3D scanning spray that does the same reflection damping and is sometimes used when scanning shiny jewelry.  An example of such a spray is shown in the link below:

https://www.laserdesign.com/3d-scan-spray/?language=en_US

impet14, 2020-11-29:
Thank you very much for your suggestion. I might try to use a linear polarizer.

MartyG-RealSense, 2020-12-05:
Hi @impet14   Do you have an update on this case that you can provide please?   Thanks!

closed 2020-12-12 by MartyG-RealSense