realsenseai/realsense-ros #1529: Huge drift of IMU's

zapaishchykova, 2020-11-28:
Hello! I am trying to use it for realtime scanning of the environment with Ubuntu 18 + D435i and no matter how slow I move, I get drift of IMU's that is mentioned on the main page. Any suggestions which configs to change or what to try to make it more robust?

MartyG-RealSense, 2020-11-28:
Hi @zapaishchykova   If the navigation of a robot equipped with an IMU starts to drift then it can be re-oriented with a process called **relocalization**.    If using the D435i and its IMU with ROS, the link below suggests making use of the **imu_filter_madgwick** module that is already part of the D435i SLAM example to do so,

https://answers.ros.org/question/326273/d435i-realsense-and-robot_localization/

zapaishchykova, 2020-11-29:
What if there is no robot and environment is going to be captured with a hand, so that means I only use IMU of the D435i (so i don't have any robot localization)? I have used this link to set up https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i, and there I mentioned I have this problem with huge drift.
I don't really care about the position of the camera inside my tracked sequence, I just want to have a realtime scan of the environment.

MartyG-RealSense, 2020-11-29:
To capture the environment whilst moving around holding the camera, you would not necessarily require ROS to do that.  The commercial software **DotProduct Dot3D Pro** can do this and is compatible with D435i.

https://www.dotproduct3d.com/dot3dpro.html
https://www.youtube.com/watch?v=OxUDLWv5Poo

As well as the premium Dot3D Pro, there is also an entry-level **Dot3D Scan** version.

https://www.dotproduct3d.com/dot3dscan.html

zapaishchykova, 2020-11-29:
Are there any alternatives in open-source?

MartyG-RealSense, 2020-11-29:
Another RealSense team member has highlighed an open-source RealSense handheld scanning project based on RTABMAP that does not need an IMU.

https://hackaday.com/2020/03/31/handheld-3d-scanning-using-raspberry-pi-4-and-intel-realsense-camera/

zapaishchykova, 2020-11-29:
thank you, RTABMAP was exactly what I was looking for!

closed 2020-11-29 by zapaishchykova