realsenseai/realsense-ros #1530: Failed to load plugin image_transport/compressed_pub

iggyrrieta, 2020-11-29:
- I succesfully compiled [this version](https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.1) following [this instructions](https://github.com/IntelRealSense/realsense-ros/blob/3.1.1/README.md). I am using `eloquent` branch.

- The **librealsense** library compiled on my computer is **v2.38.1**

- I am using **Ubuntu 20.04** and **ROS2 foxy**

When running launch file `rs.launch.py` or `rs.d400_and_t265.launch`, d435 camera get the following error:
```bash
[ERROR] [1606663452.863241363] [d435.camera]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
```

This is the only error and I get this message after that:
```bash
[realsense2_camera_node-2] [INFO] [1606663452.981769735] [RealSenseCameraNode]: RealSense Node Is Up!
```

I already checked #1337 but i am using a version ahead (3.1.1)

Thanks in advance!

olivier-haddad, 2020-11-29:
I have been battling the same problem for the last few days using the following versions:

- Ubuntu Focal 20.04, ROS2 Foxy

- The librealsense library from released binaries: librealsense2/focal,now 2.40.0-0~realsense0.3815 amd64 [installed,automatic]

- I tried multiple realsense-ros versions and tags from the "eloquent" branch: 3.1.1, e0e6f20951a936e8cd3badc602f3bd47ea590730 (oct. 8th),  281ae930236e51b9dd664b2b24349063bd01fc01 (Nov 29th)

These errors appear:
`[realsense2_camera_node-1] [INFO] [1606675274.202987526] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [ERROR] [1606675274.209621847] [camera1.cam1]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [ERROR] [1606675274.210267949] [camera1.cam1]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [ERROR] [1606675274.210814215] [camera1.cam1]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1606675274.211454092] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1] [INFO] [1606675274.235913062] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1606675274.268777292] [RealSenseCameraNode]: RealSense Node Is Up!
`

The topics are created but the cameraInfo topic shows dimensions 0,0.

RealSenseSupport, 2020-12-02:
@olivier-haddad Could you please check if https://github.com/cipherdev/src/issues/7 helps on your issue? And please note that ROS2 foxy not officially supported for RealSense ROS wrapper 2 yet. So you might get some issues. Sorry for that.

iggyrrieta, 2020-12-02:
Thanks @RealSenseSupport for your help! I already had the `image_transport_plugins` installed, actually reading the error @olivier-haddad and I have, I think the plugin is found but and error pops up when using a parameter called `format`

```bash
[ERROR] [1606663452.863241363] [d435.camera]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
```

olivier-haddad, 2020-12-02:
@RealSenseSupport Thank you for your answer.

Indeed, as @iggyrrieta assumed, I have the following installed:

- ros-foxy-image-transport-plugins/focal,now 2.3.0-1focal.20201103.161741 amd64 [installed]
- ros-foxy-compressed-image-transport/focal,now 2.3.0-1focal.20201103.161309 amd64 [installed]
- ros-foxy-camera-info-manager/focal,now 2.3.0-1focal.20201103.153418 amd64 [installed,automatic]

I know that foxy is not yet supported but as Eloquent has already reached EOL (Nov 2020), and Foxy being an LTS released nearly half a year ago, we're still hoping to see a foxy branch soon.

iggyrrieta, 2020-12-02:
I've been able to test the `eloquent branch` using `ROS2 eloquent` on Ubuntu 18.04. It works as expected. So I'll use it this way until it works on Foxy.

So I am closing this issue because as @olivier-haddad  said:
`I know that foxy is not yet supported but as Eloquent has already reached EOL (Nov 2020), and Foxy being an LTS released nearly half a year ago, **we're still hoping to see a foxy branch soon**.`

And this Foxy-Ubuntu20.04 implementation has been already asked on other issues.

closed 2020-12-02 by iggyrrieta

AlexKaravaev, 2020-12-02:
The problem here is not in realsense package. It's reported error in image_transport_plugins. If you apply this [merge request](https://github.com/ros-perception/image_transport_plugins/pull/58) and compile from source, mentioned above errors will disappear.

olivier-haddad, 2020-12-02:
@AlexKaravaev As it works with the eloquent branch, is this a new bug that was introduced in image-transport-plugins in foxy?

AlexKaravaev, 2020-12-02:
@olivier-haddad Probably yes. Maybe some parameter declaring mechanism changed in foxy compared to eloquent, I don't know for sure

olivier-haddad, 2020-12-07:
@AlexKaravaev Thanks. It indeed did the trick. It is funny that the PR is pending for six months already and hasn't yet been merged.