realsenseai/realsense-ros #1531: Eloquent branch Usage Instructions need be updated

RealSenseSupport, 2020-11-30:
to start the camera nodes in ROS2 eloquent branch

ros2 run realsense2_node realsense2_node 
need change to
ros2 run realsense2_camera realsense2_camera_node

MartyG-RealSense, 2020-11-30:
Thanks very much @RealSenseSupport - I wll tag this case as **Documentation** as a reminder to keep it open and tag @doronhi in on this message.

iggyrrieta, 2020-12-02:
Not only in `README.md`, the launch file `rs.d400_and_t265.launch.py` on `eloquent branch` have the same issue.

At [rs.d400_and_t265.launch.py](https://github.com/IntelRealSense/realsense-ros/blob/eloquent/realsense2_camera/launch/rs.d400_and_t265.launch.py) the packages are also pointing to `realsense2_node` 

**Changes**

`package='realsense2_node'` --> `package=realsense2_camera`
`node_executable='realsense2_node'` --> `node_executable=realsense2_camera_node`

MartyG-RealSense, 2020-12-02:
Thanks very much for the report @iggyrrieta

doronhi, 2020-12-07:
Thanks. I hope to finally attend this issue very shortly.

MartyG-RealSense, 2020-12-14:
Adding a note to keep this case open for a further time period.

doronhi, 2020-12-14:
I see now that I made that mistake also in the new [Release 3.1.2](https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.2)
I fixed it now in the [eloquent branch](https://github.com/IntelRealSense/realsense-ros/blob/eloquent/README.md).

MartyG-RealSense, 2020-12-21:
Thanks very much for providing the fix @doronhi   :)

Closing this case now that the fix has been provided.

closed 2020-12-21 by MartyG-RealSense