realsenseai/realsense-ros #1532: Unable to run opensource_tracking example on Jetson Nano

arjunskumar, 2020-12-01:
While running the opensource_tracking.launch, i'm getting failed to meet update rate warning. 
Any idea how can I run RTAB on Jetson Nano

MartyG-RealSense, 2020-12-01:
Hi @arjunskumar  Which librealsense version and RealSense ROS wrapper version are you using please?   Thanks!

arjunskumar, 2020-12-01:
@MartyG-RealSense Librealsense Version 2.31 and Realsense ROS wrapper version 2.2.0

MartyG-RealSense, 2020-12-01:
It is important to match the ROS wrapper version as closely as possible with the specific librealsense SDK recommended on the Releases page for the wrapper.  For ROS wrapper 2.2.0, librealsense **2.40.0** should be used with it.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.20

Once librealsense 2.40.0 is installed, the RealSense ROS wrapper should be rebuilt.

MartyG-RealSense, 2020-12-08:
Hi @arjunskumar   Do you require further assistance with this case, please?   Thanks!

arjunskumar, 2020-12-10:
@MartyG-RealSense I have tested wit Jetson TX2 by building realsense SDK from source, and its working. I will update you once I  switch to Jetson Nano.

MartyG-RealSense, 2020-12-10:
Thanks very much for the update @arjunskumar - I look forward to your next update!

MartyG-RealSense, 2020-12-17:
Adding a note to keep this case open for a further time period.

arjunskumar, 2020-12-23:
@MartyG-RealSense The error was because of opencv installation that comes with Jetson Nano. I had to reinstall the opencv and now its working.

closed 2020-12-23 by arjunskumar

MartyG-RealSense, 2020-12-23:
Great to hear that you solved the problem - thanks very much for the update!