realsenseai/realsense-ros #1533: Hardware Notification:Depth stream start failure,1.60683e+12,Error,Hardware Error

Steven-ZY, 2020-12-01:
`roslaunch realsense2_camera rs_camera.launch`
There is an error:
"Hardware Notification:Depth stream start failure,1.60683e+12,Error,Hardware Error"

How to solve it? Thanks in advance.

MartyG-RealSense, 2020-12-01:
Hi @Steven-ZY  As a starting point in this case, could you try adding a reset command to the end of your roslaunch instruction to reset the camera at launch, please?

**initial_reset:=true**

Steven-ZY, 2020-12-01:
@MartyG-RealSense Thanks. But even I follow your instructions, the same problem wasn't solved. And I update the issue, which is more detailed.

Steven-ZY, 2020-12-01:
@MartyG-RealSense 
` 01/12 21:38:31,523 ERROR [140131956680448] (ds5-options.cpp:86) Asic Temperature value is not valid!
 01/12 21:38:32,523 ERROR [140131956680448] (ds5-options.cpp:86) Asic Temperature value is not valid!
 01/12 21:38:33,524 ERROR [140131956680448] (ds5-options.cpp:86) Asic Temperature value is not valid!
`

MartyG-RealSense, 2020-12-01:
Is the camera ble to stream depth correctly in the **RealSense Viewer** program?

Steven-ZY, 2020-12-01:
@MartyG-RealSense Yes, it's OK in **realsense viewer** program.

MartyG-RealSense, 2020-12-01:
That is a positive sign as it shows that there is not a hardware problem with the camera.

I'm not convinced that the Motion Module Force Pause warning would make a difference to the launch.

**Asic temperature is not valid** could suggest a lengthy timeout (lack of arriving data) from the camera if the Asic temperature is checked every 10 seconds and the warning has occurred three times in the log.

https://github.com/IntelRealSense/realsense-ros/issues/894

Can you check if the depth topic is publishing data despite the depth stream start error please:

**rostopic echo /camera/depth/image_rect_raw**

MartyG-RealSense, 2020-12-08:
Hi @Steven-ZY  Do you require further assistance with this case, please?   Thanks!

Steven-ZY, 2020-12-08:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    No, thanks. I figured it out.
        
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                陈志远
                            
                            
                                
                            
                    
                        
                            
                                    清华大学
                                    
                                    chenzy19@mails.tsinghua.edu.cn
                                    
                                    
                                    
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

MartyG-RealSense, 2020-12-08:
Okay, great to hear - thanks very much for the update!

closed 2020-12-08 by MartyG-RealSense

Tal-seven, 2021-01-03:
@Steven-ZY would be helpful if you mention how you solved the above mentioned issue as we face a similar one from time to time on some of our robots

Steven-ZY, 2021-01-05:
@Tal-seven Hey, bro. You may use the keyboard combination Super+O to toggle the rotation lock ON/OFF.

brentalanmiller, 2021-01-08:
I'm running into the exact same problem and I'm not sure how work around it.

The issue seems to come up when I add enable the gyro and accel outputs.

If I run `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true` it works fine.

But with `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true enable_gyro:=true enable_accel:=true` I'll get the error every time.

I can kind of work around it if I remove the `initial_reset:=true`, launch it once without the gyro and accel, and then launch it again with those flags, but that seems to only work about 50% of the time.

My full output when I see the error:
```
$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true enable_gyro:=true enable_accel:=true
... logging to /home/bmiller/.ros/log/85ea8ec8-51f4-11eb-8b49-000c29657ffb/roslaunch-hawking-9024.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hawking:43833/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
[... 113 lines cut ...]
 08/01 13:00:31,347 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,399 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,450 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1610139631.450894670]: Hardware Notification:Motion Module failure,1.61014e+12,Error,Hardware Error
 08/01 13:00:31,502 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,552 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,604 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,656 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,707 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,757 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 13:00:31,809 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 08/01 13:00:32,070 WARNING [140208966317824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1610139632.454708054]: Hardware Notification:Depth stream start failure,1.61014e+12,Error,Hardware Error
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2021-01-08:
Hi @brentalanmiller   Do you see any improvement in reliability if you add the command below to the end of your roslaunch instruction?

**unite_imu_method:=linear_interpolation**

brentalanmiller, 2021-01-08:
Hi @MartyG-RealSense  No, same error:

```
$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation
... logging to /home/bmiller/.ros/log/55688310-5201-11eb-8886-000c29657ffb/roslaunch-hawking-6134.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hawking:44661/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 105 lines cut ...]
 08/01 14:32:13,508 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,559 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1610145133.603198618]: Hardware Notification:Motion Module failure,1.61015e+12,Error,Hardware Error
 08/01 14:32:13,609 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,661 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,713 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,765 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,815 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,867 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,919 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/01 14:32:13,970 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 08/01 14:32:14,228 WARNING [139948153894656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1610145134.604381613]: Hardware Notification:Depth stream start failure,1.61015e+12,Error,Hardware Error
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2021-01-08:
Your librealsense and ROS wrapper versions are matched correctly (2.40.0 SDK and 2.2.20 ROS wrapper).

As you do not have problems when the IMU topics are not enabled, that would suggest that the librealsense and ROS wrapper installations are okay.

The warning **control_transfer returned** suggests a serious communication problem with the device if the warning is generating continuously instead of just several times during launch, or once or twice per minute after launch.

I note that your log is reporting that the IMU is not calibrated.

**08/01 14:32:12,699 WARNING [139948410152704] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.**

Have you performed a calibration of your D455's IMU yet please?

Tal-seven, 2021-01-09:
@brentalanmiller Please do try what @MartyG-RealSense has suggested. Also have you updated the firmware of your realsense recently. Try downgrading a revision and see if the error goes away, it helped me.

brentalanmiller, 2021-01-09:
Thanks for the help @Tal-seven!, I downgraded to 05.12.09.00 and that seems to have fixed the problem. I still get the warning `Hardware Notification:Motion Module failure,1.61023e+12,Error,Hardware Error`, but the IMU data seems to be coming in fine. Worth noting that I have also seen a similar warning in both the ROS driver and the RealSense Viewer on both 05.12.09.00 and 05.12.10.00.

But to @MartyG-RealSense's points, I also tried the following before downgrading:

- I ran the calibration; this didn't seem to have any affect, but I no longer get the warning about the IMU being not calibrated.

- As mentioned above, I do get a warning in the RealSense Viewer when I enable the motion module, yet the data seems to be coming in just fine. I have not seen any sort of failure or warning regarding the depth module in the viewer though:
[image]

Thanks to both of you for your help.

MartyG-RealSense, 2021-01-10:
Thanks so much @Tal-seven for your help!

Hi @brentalanmiller   Great news that you had successful results.  Yes, the **Motion Module Failure** warning does not prevent the IMU from working normally in the Viewer after the message has occurred.

LeoWang19, 2022-02-16:
I had the same error with the depth module failure.

I believe it is because of the **resolution** with the depth sensor + its **fps**.

I have tried different resolution and fps combinations. no failure for depth sensor if:
     640*480 + 60 fps (30fps causes failure)
      848*480 + 30fps

LeoWang19, 2022-02-16:
note the previous trial is on D435.

rakshith95, 2022-03-24:
@MartyG-RealSense I get the same error: "[ WARN] [1648125842.425313765]: Hardware Notification:Depth stream start failure,1.64813e+12,Error,Hardware Error" when I start with the depth stream as 640x480, but not in, say, 848x480. 
The depth topics seem to publish fine.

What could be the reason for this warning?

ozilxu, 2022-07-26:
Hi @brentalanmiller @Tal-seven @rakshith95 @MartyG-RealSense @wx1909 Can you help me check my problem?

I have similar issue. And I am using Ubuntu 18.04.6 in VMware. 

The **D415 camera** works fine with real-sense viewer but it fails to open in terminal with ros and rviz: 

```
roslaunch realsense2_camera rs_camera.launch
... logging to /home/xxuaz/.ros/log/60930144-0cdf-11ed-9172-000c29e8b8cd/roslaunch-xxuaz-virtual-machine-14486.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xxuaz-virtual-machine:43979/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 91 lines cut ...]
[ INFO] [1658838911.050161856]: RealSense Node Is Up!
[ WARN] [1658838911.098796119]: 
 26/07 20:35:11,101 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,153 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,206 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,259 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,311 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,363 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,415 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,467 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,519 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:11,572 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[ WARN] [1658838912.629139610]: Hardware Notification:Depth stream start failure,1.65884e+12,Error,Hardware Error
 26/07 20:35:12,753 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/07 20:35:12,910 WARNING [139859553408768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1658838914.638409090]: Hardware Notification:Depth stream start failure,1.65884e+12,Error,Hardware Error
```

tulingcheng86, 2024-05-06:
> Hi @Steven-ZY As a starting point in this case, could you try adding a reset command to the end of your roslaunch instruction to reset the camera at launch, please?
> 
> **initial_reset:=true**

thanks!That's very good for me

Wong-Xian, 2024-10-31:
I concured the same issue, solving by downgrade the Firm Ware from 5.16.01 to 5.15.1

here is the log before downgrading, when executing 'roslaunch realsense2_camera rs_camera.launch' :
`
[ INFO] [1730349401.836960809]: Initializing nodelet with 4 worker threads.
[ INFO] [1730349401.919228732]: RealSense ROS v2.3.2
[ INFO] [1730349401.919270334]: Built with LibRealSense v2.50.0
[ INFO] [1730349401.919284623]: Running with LibRealSense v2.50.0
[ INFO] [1730349401.976011031]:  
[ INFO] [1730349402.197326730]: Device with serial number 043422251011 was found.

[ INFO] [1730349402.197433271]: Device with physical ID 2-4-10 was found.
[ INFO] [1730349402.197478980]: Device with name Intel RealSense D455 was found.
[ INFO] [1730349402.197835778]: Device with port number 2-4 was found.
[ INFO] [1730349402.197925979]: Device USB type: 3.2
[ INFO] [1730349402.199253402]: getParameters...
[ INFO] [1730349402.231408992]: setupDevice...
[ INFO] [1730349402.231526950]: JSON file is not provided
[ INFO] [1730349402.231588458]: ROS Node Namespace: camera
[ INFO] [1730349402.231647144]: Device Name: Intel RealSense D455
[ INFO] [1730349402.231703368]: Device Serial No: 043422251011
[ INFO] [1730349402.231759672]: Device physical port: 2-4-10
[ INFO] [1730349402.231815918]: Device FW version: 05.16.00.01
[ INFO] [1730349402.231871899]: Device Product ID: 0x0B5C
[ INFO] [1730349402.231927520]: Enable PointCloud: Off
[ INFO] [1730349402.231982839]: Align Depth: On
[ INFO] [1730349402.232038162]: Sync Mode: On
[ INFO] [1730349402.232117063]: Device Sensors: 
[ INFO] [1730349402.317747108]: Stereo Module was found.
[ INFO] [1730349402.335366646]: RGB Camera was found.
[ INFO] [1730349402.335650559]: Motion Module was found.
[ INFO] [1730349402.335916826]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1730349402.336429978]: num_filters: 1
[ INFO] [1730349402.336489329]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1730349405.224567633]: Done Setting Dynamic reconfig parameters.
[ INFO] [1730349405.225047127]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1730349405.225483480]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1730349405.226532748]: gyro stream is enabled - fps: 200
[ INFO] [1730349405.226563703]: setupPublishers...
[ INFO] [1730349405.229116830]: Expected frequency for depth = 30.00000
[ INFO] [1730349405.232533705]: Expected frequency for color = 30.00000
[ INFO] [1730349405.234201520]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1730349405.237278432]: setupStreams...
 31/10 12:36:45,401 WARNING [140558091228928] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 31/10 12:36:45,693 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 31/10 12:36:45,794 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:45,845 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:45,895 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1730349405.945661127]: SELECTED BASE:Depth, 0
[ INFO] [1730349405.953794214]: RealSense Node Is Up!
[ WARN] [1730349406.037259080]: 
 31/10 12:36:46,037 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,088 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,138 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,188 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,238 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,289 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,339 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,390 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,440 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:46,459 ERROR [140557015832320] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
[... cut ...]
 31/10 12:36:47,860 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:36:48,718 WARNING [140558066050816] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
`

I can't fetch msgs using 'rostopic echo /camera/gyro/sample'. and even after killing the rs_camera.launch process, realsense-viewer can't recieve the msgs from imu.

but after downgrading, everything works fine.

MartyG-RealSense, 2024-10-31:
Thanks so much @Wong-Xian  for sharing what worked for you with your own issue!