realsenseai/realsense-ros #1534: Realsense d435i depth cloud not straight against wall.

ghokulji, 2020-12-01:
Hello,

I am using a realsense d435i, when I place the d435i parallel to the wall, ideally the depth cloud should be straight against the wall. Whereas in my case the depth cloud is curved and not straight.

Top View

[image]

Side View
[image]

In the above images the scan is shown in the form of red points, if you comapre the scan points with the depth you can clearly see that the depth is curved. This is causing problems while I am mapping.

This is the output of the roslaunch file.
****************************************************
PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 15
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: True
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: linear_interpolation
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://192.168.10.201:11311

process[camera/realsense2_camera_manager-1]: started with pid [20349]
process[camera/realsense2_camera-2]: started with pid [20350]
[ INFO] [1606827937.232148916]: Initializing nodelet with 6 worker threads.
[ INFO] [1606827937.528625753]: RealSense ROS v2.2.17
[ INFO] [1606827937.528718342]: Built with LibRealSense v2.39.0
[ INFO] [1606827937.528766386]: Running with LibRealSense v2.39.0
[ INFO] [1606827937.572033364]:  
 01/12 18:35:37,587 WARNING [546836558208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1606827937.796537079]: Device with serial number 923322072183 was found.

[ INFO] [1606827937.796654336]: Device with physical ID 2-1.2-22 was found.
[ INFO] [1606827937.796702256]: Device with name Intel RealSense D435I was found.
[ INFO] [1606827937.797920413]: Device with port number 2-1.2 was found.
[ INFO] [1606827937.798036711]: Resetting device...
[ INFO] [1606827943.857535701]:  
[ INFO] [1606827944.083157348]: Device with serial number 923322072183 was found.

[ INFO] [1606827944.083339776]: Device with physical ID 2-1.2-23 was found.
[ INFO] [1606827944.083383486]: Device with name Intel RealSense D435I was found.
[ INFO] [1606827944.084964290]: Device with port number 2-1.2 was found.
[ INFO] [1606827944.093747109]: getParameters...
[ INFO] [1606827944.220881543]: setupDevice...
[ INFO] [1606827944.220972840]: JSON file is not provided
[ INFO] [1606827944.221003134]: ROS Node Namespace: camera
[ INFO] [1606827944.221034344]: Device Name: Intel RealSense D435I
[ INFO] [1606827944.221058721]: Device Serial No: 923322072183
[ INFO] [1606827944.221083931]: Device physical port: 2-1.2-23
[ INFO] [1606827944.221113682]: Device FW version: 05.12.08.200
[ INFO] [1606827944.221145809]: Device Product ID: 0x0B3A
[ INFO] [1606827944.221169394]: Enable PointCloud: On
[ INFO] [1606827944.221202104]: Align Depth: On
[ INFO] [1606827944.221224231]: Sync Mode: On
[ INFO] [1606827944.221323195]: Device Sensors: 
[ INFO] [1606827944.260093484]: Stereo Module was found.
[ INFO] [1606827944.290057827]: RGB Camera was found.
[ INFO] [1606827944.290793371]: Motion Module was found.
[ INFO] [1606827944.290902294]: Add Filter: pointcloud
[ INFO] [1606827944.292002652]: num_filters: 1
[ INFO] [1606827944.292094741]: Setting Dynamic reconfig parameters.
[ INFO] [1606827944.926065016]: Done Setting Dynamic reconfig parameters.
[ INFO] [1606827944.927135914]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1606827944.928250939]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1606827944.930130219]: setupPublishers...
[ INFO] [1606827944.937532914]: Expected frequency for depth = 15.00000
[ INFO] [1606827944.999258458]: Expected frequency for color = 15.00000
[ INFO] [1606827945.032245482]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1606827945.076333757]: Start publisher IMU
[ INFO] [1606827945.079355897]: setupStreams...
[ INFO] [1606827945.093766138]: insert Depth to Stereo Module
[ INFO] [1606827945.094079157]: insert Color to RGB Camera
[ INFO] [1606827945.094322588]: insert Gyro to Motion Module
[ INFO] [1606827945.094538434]: insert Accel to Motion Module
 01/12 18:35:45,111 WARNING [546870129024] (ds5-motion.cpp:480) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 01/12 18:35:45,213 WARNING [546685555072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1606827945.268357298]: SELECTED BASE:Depth, 0
[ INFO] [1606827945.303959730]: RealSense Node Is Up!
**********************************************************************************

Any suggestions as to how to rectify the depth cloud and align it with the wall in a straight manner?

MartyG-RealSense, 2020-12-01:
Hi @ghokulji   Would it be possible please to check the wall in the **RealSense Viewer** program?  I ask this because I have seen cases of the same curved wall effect in the **RealSense Viewer** and the **Depth Quality Tool** programs, so it may not be related to the RealSense ROS wrapper specifically.  Below are links to cases where this issue has occurred in those programs.

https://github.com/IntelRealSense/librealsense/issues/5435#issuecomment-622520372

https://support.intelrealsense.com/hc/en-us/community/posts/360050158434-Curvature-on-flat-surface-wall

Plain walls with low texture or no texture can be difficult for the camera to read depth detail from.  Usually, the way to deal with this would be to provide strong illumination to bring out the texture of the wall, or to project a semi-random dot pattern from the camera's projector onto the wall.  The camera can use the dots projected onto the wall surface as a texture source to analyse for depth.  In the absence of the dot pattern projection or strong illumination, the image may be erratic.

[image]

ghokulji, 2020-12-02:
Hello Marty,

So I connected the camera to the realsense-viewer and changed the setting to a wall with texture and color. Here are the results.

This is the wall, the 2D image

[image]

This is the top view of the 3D

[image]

[image]

From my observation it seems that the ends of the image are curved a lot and the middle part is curved in. It seems like the depth image is projected behind the wall. Also, about projecting a semi-random dot pattern which you had mentioned, how to enable that in ROS? 

Can you suggest any process that I should go through to fix the curved depth image?

MartyG-RealSense, 2020-12-02:
The inward bending is characteristic of a depth distortion effect called a **pincushion distortion** (outward bulging is a barrel distortion).  Dorodnic the RealSense SDK Manager suggests calibrating the camera with the **Dynamic Calibrator** tool. 

https://github.com/IntelRealSense/librealsense/issues/4939#issuecomment-535458739

The tool can be downloaded for Windows from the link below

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool?v=t

If you are using Linux, you can install the tool for Linux using instructions on page 14 onwards of the Dynamic Calibration User Guide for the tool.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

ghokulji, 2020-12-03:
Hello,

I calibrated the d435i 3 times using the Dynamic Calibrator and still I am facing the curvature issue. I also used the depth quality tool to check if there is an issue with depth, but the depth is of correct measurement.

You can see that the cloud is curved, whereas it should be aligned with the grid

[image]

This is the realsense viewer after the dyanmic calibration.

[image]

[image]

Is there anything else I can do to rectify this? Any sort of help or suggestions will be useful.

Thanks!

MartyG-RealSense, 2020-12-03:
I performed my own check against a flat wall with two different cameras and rotated the point cloud overhead to check that the point cloud was not distorting when rotated.  The wall on my cloud remained flat from all viewing angles, suggesting that there is indeed some form of distortion occurring with your particular camera.

Could you activate the two **Infrared** streams in the Viewer please in the **2D** mode of the Viewer to see if there is any obvious problem with their images, since the depth frame on the 400 Series cameras is generated by the camera hardware from the left and right infrared frames.

ghokulji, 2020-12-04:
Hello,

So, I activated both the infrared streams in Viewer(2D mode). The images seem fine to me. Here are the images.

[image]

Can you show me an example of how your 3D output (top view) is against a wall? , it can give some perspective how the output should actually be. 

Also, if it is a camera issue, what are the steps I can take next to get it corrected?

MartyG-RealSense, 2020-12-04:
Here are front, top-down and side views of a wall taken with my D435i:

[image]

[image]

[image]

A small amount of curvature can be observed in the above images, though not nearly to the extent of yours.

I would expect problems with the left or right infrared sensors to affect the depth image, but the IR seems fine.  The depth in the Viewer is not generated by the Viewer's IR streams (since depth can be generated in the Viewer without the Viewer's IR streams being enabled), but I would expect problems with the IR imagers to manifest in the Viewer IR streams too.

Your 2D depth image of the chair in the scene with the IR dot pattern enabled actually looks quite accurate.   The dot pattern may be a contributing factor in that, as it acts as a texture soure for the camera to read depth detail from.   Could you re-test the wall on its own with the IR streams active in the Viewer to replicate the conditions of the chair scene please?

ghokulji, 2020-12-04:
Hello,

I tested against the wall, here are the results

2D viewer

[image]

3D viewer

Top View

[image]

[image]

Can this be due to some hardware issue in the camera?

MartyG-RealSense, 2020-12-04:
I will enquire to Intel for their advice.   Thanks so much for your patience.

ghokulji, 2020-12-07:
Hello Marty,

Can you advice me on what are the next steps I can take to proceed further? Also, if it is a hardware issue, can I claim the 1 year warranty.

Thanks!

MartyG-RealSense, 2020-12-07:
I have not heard back from Intel yet.  I have sent them a reminder about it.   Thanks for your patience!

ghokulji, 2020-12-07:
Okay, Thank you for the response!

jnsweets, 2020-12-08:
Hello,

I will try to help with this issue. 
In looking over the images and info exchange, it appears your camera has undergone a fairly large effective change of one or more calibration parameters. In particular, it looks like an extreme bending (or equivalent) has occurred. 

One relevant question is whether this is due to a modification or corruption of the calibration coefficients or an actual physical change to the camera. Was the performance like this when you first used it (out of the box) or did it occur afterward? If afterward, do you know what immediately preceded it - drop, shock, temp variation, attempt to re-calibrate,...?

If there's reason to believe the calibration files themselves were changed, then we could try re-loading the factory calibration to recover its original performance. If we believe the camera physically changed since its original calibration, then we can try to re-calibrate.

The signatures of curved depth against a flat wall and the invalid pixels in the 4 corners are both consistent with an Ry error (effective bending of the camera module). In this case, it looks to be ~1.5 deg which is almost 10x larger than normal variation of this parameter. 

Attached are some examples of the effect of such an error on a D435 camera placed ~1m from a flat wall. It shows the original normal performance, performance after a +1.5 deg Ry change is applied, and the recovery to near original performance after running a version of self calibration (explained more below).

You mentioned that the Z error was measured to be small, but the above type of error would also result in a very large Z error unless there was something else changed that might compensate for that. In my example, the Z is off by ~35% at 1m (measures ~650mm in center and less near edges). 

Can you describe the procedure you used to determine that the Z accuracy is OK?

You mentioned you tried dynamic calibration but it did not help. DC in principle can fix or improve this type of error, but the magnitude of the error may be too large for that to work well. There is also the possibility that it was not run properly or had some other issue. Did you run it using the scale target? This would be needed in order to correct this type of error. Did it report an error or was it "successful"?

Finally, there is a newer version of calibration that should improve or fix this type of error. It's described in this WP:

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

I can help you test this out if we find that it's warranted based on your responses to the above.

John
[Effect of Ry on Depth_D435.pptx](https://github.com/IntelRealSense/realsense-ros/files/5655968/Effect.of.Ry.on.Depth_D435.pptx)

ghokulji, 2020-12-08:
Hello John,

The issue of invalid pixels on all 4 sides was there from the first time I used the camera. I tried the dynamic calibration method many times, but there was not any significant change. 

For the z accuracy, I used the depth quality tool, I measured the depth from the wall using a measuring tape and compared this to the value given by the depth tool.  The values were almost same.
But the depth of the 3D cloud was not correct (used the inbuilt measure tool in the depth quality tool ), it was showing that the wall to be behind the ideal position.

I will try keeping the camera 1m away from the wall and test it as you have shown it in the examples.I will also try out the newer calibration method and let you know the results. Is there anything else I can do to fix this issue?

Thanks!

jnsweets, 2020-12-08:
Thanks for info. 

If this was the state of the camera out of the box, then it is defective for some reason. Either its original calibration was bad or something happened to it post factory. There are a few options, each with pros and cons:
1. You can return the camera and get a new one. This would probably be the safest option since we do not know whether the camera is stable or not. So, even if we are able to recover it, there may be continued instability in the future.
2. We can try to recover acceptable performance on this camera. There are a few approaches to this described in more detail below. 
3. Of course, if we try #2 and are not able to recover it or there are continued issues, you can always return it.

One clarification on your determination of Z accuracy - you say that the reported Z compared with GT was OK, but that the point cloud appears behind the ideal position. I don't quite understand that part - what do you mean by ideal position? That sounds like a depth error which varies across the image due to the curved depth map. Maybe there was some issue or bug with the DQT, but this sounds like a discrepancy. 

It is possible there is more than one source of error and they are mostly balancing out near the center. We would need more detailed info to figure that out.

If you would like to simply return it, we can provide instructions for that.
If you would like to try re-calibrating, I would recommend a particular set of steps using the self-calibration tool which we can walk through either via Github or a phone call. 

John

I

ghokulji, 2020-12-09:
Hello John,

I think the problem was with the calibration data parameters. I tried the factory reset option given in the calibration data, the results are very good.

[image]

[image]

[image]

Is there any reason that even after multiple attempts of using dynamic calibration the issue was not resolved, before?

I also have a couple of questions regrading calibration:
a) You had previously mentioned that heating and vibration affects performance, I am planning to use this on a mobile robot and it will be used for a prolonged period of time. How should I make sure that these factors don't affect performance?

b) In realsense-ros where can I change the parameters that are shown in the realsense-viewer?

Thanks for the help!

MartyG-RealSense, 2020-12-09:
Great news @ghokulji  that your problem has been resolved thanks to the excellent advice of @jnsweets 

The reason why repeated attempts of dynamic calibration did not correct the issue may be related to the earlier advice of @jnsweets quoted below:

_DC in principle can fix or improve this type of error, but the magnitude of the error may be too large for that to work well.  There is also the possibility that it was not run properly or had some other issue._  

Regarding your other questions:

(a)  The depth algorithm of the 400 Series cameras is robust against vibration.  If you expect to encounter severe vibration though (e.g travelling over rough ground like that on agricultural land) then you could fit vibration **damping pads** to the camera's mounting point on the robot.

If the camera's projector temperature exceeds 60 degrees C for several seconds then the value of the **Laser Power** will be halved by a laser safety mechanism in the camera firmware.  If the excessive temperature level persists then the laser will be shut off.   Reducing laser power can cause the depth image to become more sparse (characterised by holes and gaps).

Further information about temperature and reducing it is provided in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360051461313-Realsense-d415-working-temperature-and-MTBF-

(b)  RealSense ROS uses a different system of parameter control called **Dynamic Reconfigure** (also known as rqt-reconfigure).  Parameters can be changed during runtime by accessing Dynamic Reconfigure (through a GUI interface or with **rosrun** commands).  They can also be set in the launch file with **rosparams**.   In RealSense ROS, each Dynamic Reconfigure parameter has a launch file rosparam equivalent with the same name, providing flexivility about whetther to configure before or during runtime.

More information about Dynamic Reconfigure can be found at the RealSense ROS documentation lonk below:

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

jnsweets, 2020-12-09:
That's great that you were able to recover to normal performance, and a change of calibration parameters was a possible cause. Doing the reset seems to indicate that was the issue and so the fix is pretty simple.

The question then is why/how this happened. You mentioned that the poor behavior was present from the beginning (out of box). This suggests that this calibration change occurred between leaving the factory and you receiving it. I really do not know how that would happen and have never seen cases like this. It is hopefully a one-time event but I can't really say for sure.

As for why DC was unable to resolve it, there are a few possibilities, but based on your description of the issue (large depth curvature but low Z error) as well as the fact that this was apparently not a "natural" degradation but some corruption of the calibration tables, it could be that it was a pathological case beyond the bounds of DC recovery. That's my best guess based on the evidence.

Your other questions:

a) You had previously mentioned that heating and vibration affects performance, I am planning to use this on a mobile robot and it will be used for a prolonged period of time. How should I make sure that these factors don't affect performance?

This is a more involved issue since there are a number of environmental factors that can affect performance. In general, minimizing external vibrations on the camera through mounting mechanisms (isolating or damping) will help. There is some temperature dependence on performance but unless you exceed the bounds in the data sheet or are very sensitive to depth variations, it should not be a large problem. You can also run periodic performance tests after installation on the robot to monitor any changes in performance over time or conditions. That would provide some idea of the magnitude of any effects. If periodic or occasional recalibration is needed, we have some tools for that, as you know.

b) In realsense-ros where can I change the parameters that are shown in the realsense-viewer?

This is outside my area, so I'll have to look into it.

jnsweets, 2020-12-09:
It looks like MartyG just replied to your questions with some good info. Let us know if you need more help.

ghokulji, 2020-12-09:
Thanks a lot @MartyG-RealSense and @jnsweets for taking the time to help me fix the issues.
I am closing this issue.

Thanks!

closed 2020-12-09 by ghokulji