realsenseai/realsense-ros #1538: USB problems on Jetson TX2 when using both D435 and T265

mindThomas, 2020-12-02:
Hi.
I am currently try to setup both the D435 and T265 on a Jetson TX2 platform with JetPack 4.4.1 (Ubuntu 18.04 with ROS Melodic).
I have connected both the D435 and T265 to an externally powered USB 3.0 hub.
When I run the launch file for just one of the cameras everything works fine (as long as I do it as the first thing after booting the system, I.e. having tried to run both cameras).
But if I try to run the combined launch file which launches a nodelet for each camera a lot of USB errors are thrown for T265.

The errors include:
```
 02/12 00:49:42,077 ERROR [547465716096] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 02/12 00:49:42,077 ERROR [547465716096] (tm-boot.h:42) Failed to open T265 zero interface
 02/12 00:49:43,076 WARNING [547264389504] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 02/12 00:49:43,076 ERROR [547264389504] (tm-boot.h:39) Error booting T265
[ WARN] [1606841384.136777644]: No RealSense devices were found!
[ WARN] [1606841385.135438922]: No RealSense devices were found!
 02/12 00:49:51,220 ERROR [547264389504] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 02/12 00:49:51,221 ERROR [547264389504] (tm-boot.h:42) Failed to open T265 zero interface
 02/12 00:49:51,224 WARNING [547465716096] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 02/12 00:49:51,224 ERROR [547465716096] (tm-boot.h:39) Error booting T265
[ WARN] [1606841393.284339551]: No RealSense devices were found!
[ WARN] [1606841393.288007180]: No RealSense devices were found!
```

And also:
```
[547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,655 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,707 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,759 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,811 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,863 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,915 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:09,967 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:10,018 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:10,070 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:10,122 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:10,173 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:10,225 WARNING [547356643712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/12 00:48:10,292 WARNING [547579638144] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1606841290.293204395]: Device 1/1 failed with exception: set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
[ERROR] [1606841290.293648153]: The requested device with device name containing t265 is NOT found. Will Try again.
 02/12 00:48:17,387 WARNING [547579638144] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 02/12 00:48:17,387 ERROR [547579638144] (tm-boot.h:39) Error booting T265
[ INFO] [1606841299.450285178]:  
 02/12 00:48:22,379 ERROR [547381821824] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
 02/12 00:48:22,380 ERROR [547579638144] (sensor.cpp:512) acquire_power failed: failed to set power state
 02/12 00:48:22,382 WARNING [547579638144] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1606841302.382658069]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1606841302.382788113]: The requested device with device name containing t265 is NOT found. Will Try again.
 02/12 00:48:29,463 WARNING [547579638144] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 02/12 00:48:29,463 ERROR [547579638144] (tm-boot.h:39) Error booting T265
[ WARN] [1606841311.516780274]: No RealSense devices were found!
 02/12 00:48:38,597 WARNING [547579638144] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 02/12 00:48:38,597 ERROR [547579638144] (tm-boot.h:39) Error booting T265
[ WARN] [1606841320.653282086]: No RealSense devices were found!
```

Any clues on what to do?

Best regards
Thomas Jespersen

kaspesi, 2020-12-13:
I also have this issue, tried adding 99-realsense-libusb.rules but still is not working.  Have you found a solution to this?

amilcarlucas, 2020-12-17:
We have the same issue, can this be voted up?

pawanw17, 2020-12-29:
I have the same issue, did anyone find a solution?

Beshario, 2021-09-07:
I am having the same issue with L515

o0BEN0O, 2021-12-17:
I have the issue with T265,
like this
 (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: Resource temporarily unavailable, err. num: 11
 16/12 19:11:48,849 ERROR [140109830334336] (tm-boot.h:39) Error booting T265

asimonov-TII, 2022-09-02:
same problem here. connected both D435 and T265 to jetson Xavier NX. cameras plugged in since boot.
running launch file gives this:

```
[ INFO] [1662136427.540904608]: Initializing nodelet with 6 worker threads.
[ INFO] [1662136427.578431628]: Initializing nodelet with 6 worker threads.
[ INFO] [1662136429.889206674]: RealSense ROS v2.3.1
[ INFO] [1662136429.889347819]: Built with LibRealSense v2.49.0
[ INFO] [1662136429.889444454]: Running with LibRealSense v2.49.0
[ INFO] [1662136430.780547200]: RealSense ROS v2.3.1
[ INFO] [1662136430.780701976]: Built with LibRealSense v2.49.0
[ INFO] [1662136430.780789715]: Running with LibRealSense v2.49.0
 02/09 16:33:50,882 ERROR [547272782208] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 02/09 16:33:50,882 ERROR [547272782208] (tm-boot.h:42) Failed to open T265 zero interface
 02/09 16:33:50,997 WARNING [547767697792] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 02/09 16:33:50,997 ERROR [547767697792] (tm-boot.h:39) Error booting T265
[ INFO] [1662136432.911466095]:  
[ INFO] [1662136433.124486820]:  
[ WARN] [1662136433.148843302]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1662136433.149043643]: The requested device with  is NOT found. Will Try again.
 02/09 16:33:53,146 ERROR [547532824960] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 02/09 16:33:53,146 ERROR [547767697792] (sensor.cpp:537) acquire_power failed: failed to set power state
 02/09 16:33:53,148 WARNING [547767697792] (rs.cpp:308) null pointer passed for argument "device"
[ INFO] [1662136433.286938423]: Device with serial number 138422071334 was found.
```

AbabalaT, 2023-03-03:
My T265 has the same problem only on Orange Pi Zero 2 which only has USB2.0 port.

clean-shero, 2023-03-05:
[image]
I have the same problem

carlosxlll, 2023-03-23:
> 
我也是这个问题，请问解决了吗

julyfire-wang, 2023-06-27:
> My T265 has the same problem only on Orange Pi Zero 2 which only has USB2.0 port But D435i works well the T265 works well at USB2.1 on mt laptop
> 
> bulk_transfer returned error endpoint:0x1 Resource temporarily unavailable err num 11 error booting T265

I met the same quesetion. Have you resolved the issue？

AbabalaT, 2023-06-27:
> I met the same quesetion. Have you resolved the issue？

I have to use the usb 3.0 on rasberryPi4B instead. I don't know if any usb3.0 works.

closed 2026-10-23 by MartyG-RealSense