realsenseai/realsense-ros #1539: Is the package also released for Ros2 dashing?

Schloern93, 2020-12-02:


MartyG-RealSense, 2020-12-02:
Hi @Schloern93   The RealSense-ROS ROS2 wrapper currently officially supports Eloquent, with Foxy support being a future goal.

https://github.com/IntelRealSense/librealsense/issues/5825#issuecomment-669031717

There is a ROS2 Dashing wrapper on the main **Intel** GitHub if you are unable to update to Eloquent.  You should be aware though that it stopped being updated some months ago and is not related to the official **RealSense ROS** GitHub wrapper that supports Eloquent.  So whilst you are free to make use of the Dashing wrapper, I would strongly recommend using the current wrapper if possible.

https://github.com/intel/ros2_intel_realsense

Schloern93, 2020-12-03:
Thanks for the quick response. I will try to update my systems to eloquent

closed 2020-12-03 by Schloern93

doronhi, 2021-03-16:
I would like to draw your attention, if it is still relevant, to a new branch - [ros2](https://github.com/IntelRealSense/realsense-ros/tree/ros2) that supports Dashing, Eloquent, and Foxy.
This branch will be the active branch, starting at the next release, replacing the [foxy](https://github.com/IntelRealSense/realsense-ros/tree/foxy) branch.

MartyG-RealSense, 2021-03-16:
@doronhi   Thanks very much for the information!