realsenseai/realsense-ros #1540: Set min_distance value for L515 LiDAR

jcgarciaca, 2020-12-02:
I am using a L515 LiDAR camera with `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY` as [here](https://github.com/IntelRealSense/realsense-ros/issues/1246) and I can see the pointcloud in rviz. However, since I need to define my own `min_distance` value, I exported a JSON file with `realsense-viewer` and added its path to `json_file_path` argument but the message `Device does not support advanced settings!` is shown and of course the values are not set according with my file. How could I set my own values?

doronhi, 2020-12-03:
Which versions of ROS, librealsense and realsense2_camera do you use?
Could you add the json file you try to load?

jcgarciaca, 2020-12-03:
I am using `melodic` version with `LibRealSense v2.37.0` and `RealSense ROS v2.2.16`. The generated json file below:

```
{
    "Ambient Light": 1,
    "Apd Temperature": -9999,
    "Confidence Threshold": 1,
    "Depth Offset": 4.5,
    "Depth Units": 0.000250000011874363,
    "Error Polling Enabled": 1,
    "Frames Queue Size": 16,
    "Freefall Detection Enabled": 1,
    "Global Time Enabled": 0.0,
    "Inter Cam Sync Mode": 0.0,
    "Invalidation Bypass": 0.0,
    "Laser Power": 33,
    "Lld Temperature": 25.1394882202148,
    "Ma Temperature": 25.067985534668,
    "Mc Temperature": 24.8353538513184,
    "Min Distance": 0.0,
    "Noise Filtering": 2,
    "Post Processing Sharpening": 1,
    "Pre Processing Sharpening": 4,
    "Receiver Gain": 9,
    "Sensor Mode": 0.0,
    "Visual Preset": 0.0,
    "Zero Order Enabled": 0.0,
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "480",
    "stream-ir-format": "Y8",
    "stream-width": "640"
}
```
Since I just need to set custom values for `min_distance`, `pre_processing_sharpening` and `noise_filtering`, I defined a `json` file with just these parameters, without successful results:

```
{
    "Min Distance": 0.0,
    "Noise Filtering": 2,
    "Pre Processing Sharpening": 4,
} 
```

doronhi, 2020-12-06:
The issue regarding the message "Device does not support advanced settings!" is a bug. I fixed that in #1548.
Can you build and check that PR? I didn't test for the "Min Distance" or any other parameter so I would really appreciate your feedback.

jcgarciaca, 2020-12-17:
It worked.
Thank you

closed 2020-12-17 by jcgarciaca