realsenseai/realsense-ros #1541: I want to pass additional custom parameters in the launch file.

Hiroaki-K4, 2020-12-03:
Hi. Thank you share realsense-ros repository.

I am using realsense2_camera node now. I want to pass additional custom parameters in the launch file.
For example, I added below line to rs_camera.launch and nodelet.launch.xml.
` <arg name="new_parameters"   default="abc"/>
`
But I couldn't receive parameter in base_realsense_node.cpp.
would you let me know the solution? Thanks.

RealSenseSupport, 2020-12-07:
@Hiroaki-K4  Could you please tell which ROS wrapper version and librealsense version you used? What's the camera did you use for the testing? Thanks!

Hiroaki-K4, 2020-12-07:
- ROS wrappter 2.2.17
- librealsense 2.38.1
- test camera realsense D455

Thanks!

RealSenseSupport, 2020-12-08:
@Hiroaki-K4 Did you also add the type definition in https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/includes/nodelet.launch.xml#L185 
After added the new parameter in launch file and nodelet.launch.xml, I can see the new parameter with "rosparam list". Can you see the parameter with "rosparam list"? Looking forward to your reply. Thanks!

Hiroaki-K4, 2020-12-08:
Yes, I added. But I can't see my parameter with "rosparam list". 
my custom rs_camera.launch is below. I want to add image_dir parameter.
```rs_camera.launch
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="1280"/>
  <arg name="depth_height"        default="720"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="1280"/>
  <arg name="color_height"        default="720"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="5"/>
  <arg name="depth_fps"           default="5"/>
  <arg name="infra_fps"           default="5"/>
  <arg name="color_fps"           default="5"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="image_dir"                 default="/home/nvidia/data/output_images"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>
[... 198 lines cut ...]
<param name="pose_optical_frame_id" type="str" value="$(arg pose_optical_frame_id)"/>
<param name="aligned_depth_to_color_frame_id" type="str" value="$(arg aligned_depth_to_color_frame_id)"/>
<param name="aligned_depth_to_infra1_frame_id" type="str" value="$(arg aligned_depth_to_infra1_frame_id)"/>
<param name="aligned_depth_to_infra2_frame_id" type="str" value="$(arg aligned_depth_to_infra2_frame_id)"/>
<param name="aligned_depth_to_fisheye_frame_id" type="str" value="$(arg aligned_depth_to_fisheye_frame_id)"/>
<param name="aligned_depth_to_fisheye1_frame_id" type="str" value="$(arg aligned_depth_to_fisheye1_frame_id)"/>
<param name="aligned_depth_to_fisheye2_frame_id" type="str" value="$(arg aligned_depth_to_fisheye2_frame_id)"/>
<param name="publish_tf" type="bool" value="$(arg publish_tf)"/>
<param name="tf_publish_rate" type="double" value="$(arg tf_publish_rate)"/>
<param name="odom_frame_id" type="str" value="$(arg odom_frame_id)"/>
<param name="topic_odom_in" type="str" value="$(arg topic_odom_in)"/>
<param name="calib_odom_file" type="str" value="$(arg calib_odom_file)"/>
<param name="publish_odom_tf" type="bool" value="$(arg publish_odom_tf)"/>
<param name="filters" type="str" value="$(arg filters)"/>
<param name="clip_distance" type="double" value="$(arg clip_distance)"/>
<param name="linear_accel_cov" type="double" value="$(arg linear_accel_cov)"/>
<param name="initial_reset" type="bool" value="$(arg initial_reset)"/>
<param name="unite_imu_method" type="str" value="$(arg unite_imu_method)"/>
<param name="allow_no_texture_points" type="bool" value="$(arg allow_no_texture_points)"/>
<param name="image_dir" type="str" value="${arg image_dir}"/>
</node>
</launch>
```

Thanks!

RealSenseSupport, 2020-12-09:
@Hiroaki-K4 Did you try to run _catkin_make_ after the change?

Hiroaki-K4, 2020-12-14:
Yes,  I did.
But I can't get image_dir information.

RealSenseSupport, 2020-12-16:
@Hiroaki-K4  After checked your code, I think you should modify below two lines. You should use "()" instead of "{}".

In rs_camera.launch, modify below line.
<arg name="image_dir"                value="${arg image_dir}" />   
==> <arg name="image_dir"                value="$(arg image_dir)" />

In nodelet.launch.xml, modify below line.
<param name="image_dir" type="str" value="${arg image_dir}"/>  
==> <param name="image_dir" type="str" value="$(arg image_dir)"/>

Hiroaki-K4, 2020-12-21:
Thank you for reply.
I solved. I don't use private nodehandle in my script.

closed 2020-12-21 by Hiroaki-K4