realsenseai/realsense-ros #1543: Difference between realsense ros and realsense sdk

chillcloud-dev, 2020-12-03:
Hello,

I am using realsense sdk for my KI application and had to switch to realsense-ros, since we want to decentralize the camera (we using D415 / D435 ). I used the realsense SDK to pull the RGB frames to feed into my KI. 

Now we run rs_camera.launch and I subscribe to the topic to get my desired frames. 

Since then, I noticed at my KI application that the performance dropped significantly. 
I double checked with my realsense version and it was performing like it used to. The only difference was how i got my RGB frames ( now with realsense-ros ).

I checked the rgb frame data and compared them and noticed that there are major difference in the rgb data itself.

I tested my assumption by having a static environment and captured several frames with the same light conditions, subtracted the pixel values from one version from another  (for several frames) and found out that the difference of the pixel values between the realsense and realsense-ros version was sometimes 50 up to 108. (I expected some difference because of noise values around 5 or 10 but not 108 ! )

I plotted the histogram for the rgb channels from this subtraction. I also did the same process for the frames from the same data source.

Could you explain why there is a change between the rgb frames and how we can close the gap between these two versions. 
Is there a configuration in the launch file that I need to change ? Or do I have to live with this data quality drop ?

Histogram realsense sdk vs ros: 
[image]

Histogram between frames obtained by realsense sdk :
[image]

Histogram between frames obtained by realsense-ros :
[image]

MartyG-RealSense, 2020-12-03:
The RealSense ROS wrapper (realsense2_camera) is a ROS [package ]([url](http://wiki.ros.org/Packages)) that provides a ROS node for accessing the camera.  The RealSense ROS wrapper adheres to ROS standards to provide compatibility with ROS.  This means that sometimes functions may be handled differently than librealsense would handle them.

Could you check the color height and width values in your roslaunch log please that displays the details of the camera configuration after roslaunch is initiated and confirm whether the resolution configured for color matches the resolution you used in librealsense, please?

For example, if the roslaunch file was configured to set color to 640x480 then the log details should look like this:

 **/camera/realsense2_camera/color_height: 480
/camera/realsense2_camera/color_width: 640**

chillcloud-dev, 2020-12-03:
Hello @MartyG-RealSense 

It is the same resolution and configuration as far as I know. There are also no filters enabled in the realsense-sdk as well as in the realsense-ros. 

Here is the summary of the configuration after launching realsense2_camera rs_camera.launch. 

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
[... 41 lines cut ...]
[ INFO] [1606989585.388760885]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1606989585.388885188]: setupPublishers...
[ INFO] [1606989585.390525342]: Expected frequency for depth = 30.00000
[ INFO] [1606989585.408583186]: Expected frequency for infra1 = 30.00000
[ INFO] [1606989585.419294612]: Expected frequency for infra2 = 30.00000
[ INFO] [1606989585.429539997]: Expected frequency for color = 30.00000
[ INFO] [1606989585.441152036]: setupStreams...
 03/12 10:59:45,442 WARNING [139819724171136] (backend-v4l2.cpp:1271) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[ INFO] [1606989585.472515816]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1606989585.472814156]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1606989585.473076857]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 03/12 10:59:45,473 WARNING [139819724171136] (backend-v4l2.cpp:1271) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1606989585.503464464]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1606989585.504611120]: insert Depth to Stereo Module
[ INFO] [1606989585.504761053]: insert Color to RGB Camera
[ INFO] [1606989585.504792386]: insert Infrared to Stereo Module
[ INFO] [1606989585.504811780]: insert Infrared to Stereo Module
 03/12 10:59:45,569 WARNING [139819724171136] (backend-v4l2.cpp:1271) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[ INFO] [1606989585.571829572]: num_filters: 0
[ INFO] [1606989585.571876090]: SELECTED BASE:Depth, 0
[ INFO] [1606989585.626503056]: RealSense Node Is Up!
[ INFO] [1606989585.626537546]: Setting Dynamic reconfig parameters.
[ INFO] [1606989585.792357364]: Done Setting Dynamic reconfig parameters.

Thanks !

MartyG-RealSense, 2020-12-03:
Going through your log, I note that Infra1, Infra2, Depth, Accel and Gyro are all set to True.  If your KI application does not require Depth, Infrared and IMU streams, having them all enabled could be a drag on performance.

doronhi, 2020-12-03:
This is new to me. Is it possible that image-transport compresses the images? Is "compressed-image-transport" or "theora-image-transport" installed?

chillcloud-dev, 2020-12-03:
@MartyG-RealSense  
I disabled the infrared stream as well 

Here is the summary :

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
[... 37 lines cut ...]
[ INFO] [1607003340.327234643]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1607003340.327345603]: setupPublishers...
[ INFO] [1607003340.328627308]: Expected frequency for depth = 30.00000
[ INFO] [1607003340.344268225]: Expected frequency for color = 30.00000
[ INFO] [1607003340.354434758]: setupStreams...
 03/12 14:49:00,355 WARNING [139958256555904] (backend-v4l2.cpp:1271) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[ INFO] [1607003340.385734576]: depth stream is enabled - width: 640, height: 480, fps: 30
 03/12 14:49:00,386 WARNING [139958256555904] (backend-v4l2.cpp:1271) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1607003340.415757384]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1607003340.416864617]: insert Depth to Stereo Module
[ INFO] [1607003340.417008991]: insert Color to RGB Camera
[ INFO] [1607003340.446079695]: num_filters: 0
[ INFO] [1607003340.446133064]: SELECTED BASE:Depth, 0
[ INFO] [1607003340.448899426]: RealSense Node Is Up!
[ INFO] [1607003340.448914382]: Setting Dynamic reconfig parameters.
[ INFO] [1607003340.578607281]: Done Setting Dynamic reconfig parameters.

but i still get similar histograms 

[image]

Any ideas where this difference comes from ?

@doronhi 
I am using the default image_transport to get the rgb frames. To test if there is any issue regarding the image_transport itself, I published a static image and subscribed it with ros to see if information gets lost during the way. But there were no difference.

MartyG-RealSense, 2020-12-03:
@chillcloud-dev   Since we have not discussed trying a hardware reset of the camera during launch, can you try adding the command below onto the end of your roslaunch instruction please so that a reset of the camera is performed at launch:

**initial_reset:=true**

chillcloud-dev, 2020-12-04:
@MartyG-RealSense 
it helped
thanks

closed 2020-12-04 by chillcloud-dev

MartyG-RealSense, 2020-12-04:
Great news @chillcloud-dev  - thanks for the update!