realsenseai/realsense-ros #1544: USB CAM overflow

Schloern93, 2020-12-03:
Hello,

When I start the node for my D435 camera I get the following errors.
When I start the Node with my T265 Camera everything works fine.
What is the reason for this?

[WARN] [RealSenseCameraNode]: Hardware Notification:USB CAM overflow,1.60701e+12,Error,Hardware Error
 03/12 16:07:38,005 ERROR [140539299092224] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 03/12 16:07:38,005 ERROR [140539299092224] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
^C[INFO] [rclcpp]: signal_handler(signal_value=2)

Thanks in advanced

doronhi, 2020-12-06:
Sometimes communication errors occur between the driver and the device itself. The warnings reflect those errors. These could result from temporary kernel overload, usb overload or other reasons. As long as they are not too frequent they should not affect performance.

Schloern93, 2020-12-10:
Thanks for the info. I haven't done a performance analysis yet. I will close the issue for now

closed 2020-12-10 by Schloern93