realsenseai/realsense-ros #1545: “No such device” occur when multi cameras run several hours later？

zhangcj13, 2020-12-04:
Hi:
  I am using 3 D435 cameras. When 3 cameras run 3~7hours later, the error below occured, maybe one or two camera will lost, sometimes i reconnect the camera, the device will be found, sometimes i must reboot the ubuntu system. Does anyone know how to fix the problem? thanks 

Required Info |  
-- | --
Camera Model | 3 D435
Operating System & Version | Ubuntu 16.04
ros | kinetic
SDK Version | 2.22.0

`[ INFO] [1607064245.892996309]: Handling Request: /stream?topic=/camera3/color/image_raw
 04/12 17:43:49,558 ERROR [140527591610112] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 04/12 17:43:49,560 ERROR [140527591610112] (global_timestamp_reader.cpp:152) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 04/12 17:43:49,964 ERROR [140527504717568] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 04/12 17:43:49,964 ERROR [140527504717568] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[ERROR] [1607075030.351163744]: The device has been disconnected!
 04/12 17:43:50,381 ERROR [140527625180928] (sensor.cpp:260) An error has occurred while stop_streaming()!
 04/12 17:43:54,506 ERROR [140527484839680] (types.h:243) xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device
 04/12 17:43:54,506 ERROR [140527484839680] (backend-v4l2.cpp:1230) xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device
[ WARN] [1607075034.506296034]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device,1.60708e+12,Error,Unknown Error
 04/12 17:43:54,506 ERROR [140527473972992] (types.h:243) xioctl(VIDIOC_DQBUF) failed for fd: 24 Last Error: No such device
 04/12 17:43:54,506 ERROR [140527473972992] (backend-v4l2.cpp:1230) xioctl(VIDIOC_DQBUF) failed for fd: 24 Last Error: No such device
[ WARN] [1607075034.506559139]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 24 Last Error: No such device,1.60708e+12,Error,Unknown Error
 04/12 17:43:54,508 ERROR [140527625180928] (types.h:243) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 04/12 17:43:54,508 ERROR [140527625180928] (sensor.cpp:260) An error has occurred while stop_streaming()!`

MartyG-RealSense, 2020-12-04:
Hi @zhangcj13   The librealsense SDK version **2.22.0** that you are using dates back to May 2019.  Would it be possible to use a more recent librealsense version please, and rebuild the ROS wrapper with an equally modern wrapper version that matches the choice of SDK?  

It is important to match the ROS wrapper version as closely as posible with the librealsense version recommended for that particular wrapper on the RealSense ROS **Releases** page.   If you are using librealsense 2.22.0 and wish to continue doing so then the appropriate match would be ROS wrapper version **2.2.5**.  Are you using that wrapper version, please?

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.5

The error message **xioctl(UVCIOC_CTRL_QUERY) failed Last Error** would suggest that the conflict is related to the Ubuntu kernel though.   Which kernel version are you using please?

zhangcj13, 2020-12-07:
@MartyG-RealSense  thank you for you advice，i will try to update the librealsense and realsense-ros to the lastest one. my ubuntu kernel is : linux robot 4.15.0-112-generic #113~16.04.1-Ubuntu SMP Fri Jul 10 04:37:08 UTC 2020 x86_64 GNU/Linux.
the realsense camera start up msg below may help you to know my problem.

> ... logging to /home/robot/.ros/log/dd3afeb0-3836-11eb-8359-3db8838d2839/roslaunch-robot-7281.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robot:34829/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 250
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: True
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: 15
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 15
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/enable_accel: True
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: True
 * /camera1/realsense2_camera/enable_fisheye2: True
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: True
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: pointcloud
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 480
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 640
 * /camera1/realsense2_camera/gyro_fps: 400
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera1_infra1_op...
 * /camera1/realsense2_camera/infra2_frame_id: camera1_infra2_frame
 * /camera1/realsense2_camera/infra2_optical_frame_id: camera1_infra2_op...
 * /camera1/realsense2_camera/infra_fps: 30
 * /camera1/realsense2_camera/infra_height: 480
 * /camera1/realsense2_camera/infra_width: 640
[... 418 lines cut ...]
[ INFO] [1607310404.911603303]: RGB Camera was found.
[ INFO] [1607310404.911675194]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1607310404.911778790]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1607310404.911817728]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1607310404.911842453]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1607310404.911880383]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1607310404.912205048]: setupPublishers...
[ INFO] [1607310404.916043680]: Expected frequency for depth = 15.00000
[ INFO] [1607310404.969990269]: Expected frequency for color = 15.00000
[ INFO] [1607310405.022026576]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1607310405.064984401]: setupStreams...
 07/12 11:06:45,065 WARNING [140349923002112] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Z16  ]
[ INFO] [1607310405.066301214]: depth stream is enabled - width: 640, height: 480, fps: 15
[ INFO] [1607310405.066876922]: color stream is enabled - width: 640, height: 480, fps: 15
[ INFO] [1607310405.068756893]: insert Depth to Stereo Module
[ INFO] [1607310405.068867747]: insert Color to RGB Camera
[ INFO] [1607310405.174992124]: Add Filter: pointcloud
[ INFO] [1607310405.176021868]: num_filters: 1
[... cut ...]
[ INFO] [1607310405.320117832]: Done Setting Dynamic reconfig parameters.
 07/12 11:06:45,344 WARNING [140349800445696] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1607310405.346476955]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 07/12 11:06:45,428 WARNING [140349792052992] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.

MartyG-RealSense, 2020-12-07:
Hi @zhangcj13   A RealSense team member offers advice in the link below about the  **xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device** error in the link below.

https://github.com/IntelRealSense/librealsense/issues/1889#issuecomment-426601102

You also have numerous **UVC metadata payloads not available** errors that indicate that support for hardware metadata is not enabled.  This metadata support would normaly need to be patched into the kernel if librealsense is built manually from source code.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

I would recommend dealing with the **no such device** errors first though in order to see whether the metadata errors are related to that issue.

zhangcj13, 2020-12-09:
Hi @MartyG-RealSense ,I follow your advice update the librealsense to v2.40.0 and realsense-ros to v2.2.20 , then runing  three cameras since yesterday, it does't lost any camera yet now. I thought the problem may be solved. If any other error occured, I may still need your help. At last, thank you for your reply and useful advice.

MartyG-RealSense, 2020-12-09:
Thanks very much @zhangcj13 for the update.  It's very good to hear that you have had reliable performance so far after updating your setup.

MartyG-RealSense, 2020-12-16:
Hi @zhangcj13   Do you require further assistance with this case, please?   Thanks!

zhangcj13, 2020-12-16:
@MartyG-RealSense  HI，i will colse it,tks

closed 2020-12-16 by zhangcj13