realsenseai/realsense-ros #1546: All the warnings that you get seem to be caused by the low power up board. They just mean missing frames. You can try reducing the resolution to have a better frame rate or disable some of the sensors if you don't need them (If you don't use infra2 for example, set `enable_infra2:=false`. It may save some CPU.

ranjitkathiriya, 2020-12-04:
All the warnings that you get seem to be caused by the low power up board. They just mean missing frames. You can try reducing the resolution to have a better frame rate or disable some of the sensors if you don't need them (If you don't use infra2 for example, set `enable_infra2:=false`. It may save some CPU.

_Originally posted by @doronhi in https://github.com/IntelRealSense/realsense-ros/issues/715#issuecomment-479855525_

_______________________
How to reduce resolution in ros??

MartyG-RealSense, 2020-12-04:
Hi @ranjitkathiriya   You can edit the launch file to alter the resolution of a stream type by changing the **width** and the **length** values of the depth, color, infra, etc.  For example, setting **camera/color_width** to 848 and **camera/color_length** to 480 for a color resolution of 848x480 .

ranjitkathiriya, 2020-12-10:
Thanks for Help!!

closed 2020-12-10 by ranjitkathiriya