realsenseai/realsense-ros #1547: ros2 eloquent on xavier NX L4T32.4.4 / JP4.4.1 with D455 / sdk 2.40.0 / realsens-ros 3.1.1 : negative time stamp problem rviz2 is crashing

wifibot78, 2020-12-05:
Hello,
I have increasing negative time stamps with D455 ? it makes rviz2 crash ?

**ros2 topic echo /color/image_raw  ->>** 

---
header:
  stamp:
    sec: -1080631866
    nanosec: 121030844
  frame_id: camera_color_optical_frame
height: 480
width: 640
encoding: rgb8
is_bigendian: 0
step: 1920
data: [84, 85, 80, 84, 85, 80, 90, 82, 77, 94, 86, 82, 132, 128, 121, 89, 85, 78, 79, 74, 68, 81, 76, 70, 80, 70, 78, 79, 69, 77, 105, 95, 104, 101, 91, 101, 87, 81, 82, 82, 76, 77, 86, 75, 69, 93, 82, 76, 109, 89, 84, 96, 77, 71, 88, 70, 68, 83, 66, 63, 80, 72, 70, 80, 72, 70, 127, 121, 117, 94, 89, 85, 78, 68, 75, 77, 67, 74, 78, 61, 72, 84, 67, 78, 125, 120, 116, 88, 83, 79, 76, 76, 72, 75, 75, 70, 81, 75, 78, 94, 88, 91, 86, 78, 89, 81, 74, 84, 82, 71, 81, 79, 69, 79, 88, 76, 77, 119, 107, 109, 91, 79, 81, 87, 75, 76, 89, 77, '...']
---
header:
  stamp:
    sec: -1080631866
    nanosec: 154336956
  frame_id: camera_color_optical_frame
height: 480
width: 640
encoding: rgb8
is_bigendian: 0
step: 1920
data: [86, 80, 79, 87, 81, 80, 88, 76, 77, 97, 85, 87, 127, 121, 121, 94, 88, 89, 83, 81, 93, 79, 77, 89, 83, 78, 90, 88, 83, 94, 101, 92, 97, 92, 83, 87, 84, 81, 83, 77, 74, 76, 80, 74, 75, 84, 78, 78, 100, 89, 83, 95, 84, 78, 87, 73, 71, 81, 67, 65, 88, 70, 72, 85, 67, 70, 126, 113, 101, 102, 89, 78, 95, 79, 67, 86, 71, 59, 87, 73, 71, 87, 73, 71, 125, 120, 118, 99, 93, 91, 79, 70, 70, 75, 66, 66, 81, 67, 79, 90, 77, 88, 91, 77, 93, 86, 71, 87, 83, 69, 83, 87, 73, 86, 95, 82, 86, 111, 99, 103, 92, 83, 83, 89, 80, 80, 90, 81, '...']
---

wifibot@wifibot-nx:~/dev_ws$ rviz2 
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
Aborted (core dumped)

**I have also some usb errors ? on the ros2 component ?**

05/12 12:34:24,021 WARNING [547365551216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/12 12:34:47,091 WARNING [547365551216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

**I installed realsens sdk 2.40.0 with realsens-ros wrapper 3.1.1**

rs-xxx python scripts works well and also realsense-viewer can view the stream.

For me my USB3 is OK, the ZED2 ros2 wrapper works perfectly.

Thank you for your help.

Laurent

MartyG-RealSense, 2020-12-05:
Hi @wifibot78   There was a RealSense ROS case involving negative timestamps in September 2020 on a Raspberry Pi 4 board.  It is a long discussion and there was not a clear and definitive solution, though the case may provide you with some useful insights.

https://github.com/IntelRealSense/realsense-ros/issues/1324

I note though that you are using librealsense SDK **2.40.0** with RealSense ROS2 Eloqient wrapper version 3.1.1.  It is important to match a particular ROS wrapper version as closely as possible with the librealsense version recommended on the **Release Notes** for that ROS wrapper.  In the case of 3.1.1, the most recent librealsense version recommended for it is the older **2.38.1**.

[image]

Regarding the **control_transfer returned** warnings that you encountered: if the warning is being generated continuously then it can indicate a communication problem with the device. 

If it is generated only once or twice at start-up and then does not repeat, or only appears at a rate of once or twice a minute after launch, then it is safe to ignore the warning if the launch does not exit.

wifibot78, 2020-12-07:
Hello,

Thank you for your answer, On another xavier nx I installed V2.38.1 with Eloquent and realsense-ros 3.1.1 as mentionned, and same thing ! the time stamp starts negative ? I re flashed D455 firmware with the one provided with v2.38.1 project in case it was something related to the last firmware. and yes usb errors are only at startup, all is ok after.

I have a D435 , I will make the test.

Laurent

MartyG-RealSense, 2020-12-07:
Thanks very much @wifibot78 - I look forward to hearing about the outcome of your tests.   Good luck!

wifibot78, 2020-12-08:
Hello,

finally the time stamp went positive by setting "enable_sync" to true, rviz2 is getting data and not crashing, but there is errors on 
data reception and noting is displayed when I get /camera1/color/image_raw for exemple, I will post the output I am not at office at this moment.

Best Regard,
Laurent

MartyG-RealSense, 2020-12-08:
Thanks very much for the update!   I look forward to your results later.

wifibot78, 2020-12-09:
there is the output of rviz2, if I echo the image raw topic, I have image data, all seems ok, it seems that rviz2 is filtering data ? : 

 [INFO] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1607469763.445 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1607469763.478 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1607469763.512 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1607469763.546 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1607469763.579 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1607469763.613 for reason 'Unknown'

Any idea ?

we are blocked we need to deliver multiple D455 on our robots.

thank you

MartyG-RealSense, 2020-12-09:
Thanks for the update.   Could you try adding the camera reset command below to the end of your roslaunch instruction please if you have not done so already in order to see if it improves stability?  Including the command in your roslaunch will perform a hardware reset of the camera at launch:

**initial_reset:=true**

wifibot78, 2020-12-12:
Hello,

the reset does nothing for the timestamp.

I put back sdk 2.40.0 , set initial_reset:=true to avoid negative timestamp and all is ok.

[http://www.wifibot.com/download/img-d455.png](url)

thank you

laurent

MartyG-RealSense, 2020-12-12:
Thanks for the information.   I suggested the reset to see if it stopped **[rviz]: Message Filter dropping message** from appearing.  Is everything okay now?

wifibot78, 2020-12-12:
yes I added the reset, all ok

thank you

MartyG-RealSense, 2020-12-12:
Ok, great to hear.  Please click the **Close Issue** button under the comment box if you are satisifed to close this case.   Thanks!