realsenseai/realsense-ros #1549: Package 'realsense2_node' not found

adipandas, 2020-12-06:
Hi, when I try to start the `realsense2_node`, ROS2 eloquent is not able to find it.
I am working with **RealSense d435i**.

When I try to search for available nodes, I see `realsense2_camera` package and `realsense2_camera_node`. 
I think this happens because of name conflicts in [rs.launch.py](https://github.com/IntelRealSense/realsense-ros/blob/eloquent/realsense2_camera/launch/rs.launch.py) and [rs.d400_and_t265.launch.py](https://github.com/IntelRealSense/realsense-ros/blob/eloquent/realsense2_camera/launch/rs.d400_and_t265.launch.py).

Thus, when I try `ros2 run realsense2_node realsense2_node`, it gives me `Package 'realsense2_node' not found`.
But the command `ros2 run realsense2_camera realsense2_camera_node` works.

Is this intended? Or have I missed something?

Any help/feedback appreciated.
Thank you.

iggyrrieta, 2020-12-07:
Need to update README and some lunch files #1531

MartyG-RealSense, 2020-12-07:
Thanks so much for the link @iggyrrieta 

Hi @adipandas   Please let me know whether the link solves your problem.  Thanks!

adipandas, 2020-12-08:
Hi @iggyrrieta and @MartyG-RealSense,
Thanks for the update.

Just wanted to confirm the changes.

closed 2020-12-08 by adipandas