realsenseai/realsense-ros #1553: I want to publish rgb data and depth data in one msg

Hiroaki-K4, 2020-12-08:
Hi.
I think that realsense-ros publishes rgb data and depth data in separate topics, but I want to publish a message that contains rgb data and depth in one topic.
So I have a question. How can rgb data and depth data taken at the same moment be paired? Is it a frame_id or a timestamp?
In my experiments, I feel that depth is published faster. I tried to save rgb data and depth data with frame_id and timestamp in the file name respectively. Then, the timestamp was different in the same frame_id of rgb data and depth data.
Thanks!

MartyG-RealSense, 2020-12-08:
Hi @Hiroaki-K4   The option that is closest to what you are seeking is likely to be the **align_depth** option.  The RealSense ROS documentation states that it "gathers closest frames of different sensors, infrared, color and depth, to be sent with the same timetag.  This happens automatically when such filters as pointcloud are enabled".

Although the aligned streams are not all in one topic, it should achieve the aim of synching the timestamps.  You can add align_depth to your roslaunch instruction by adding the command below onto the end of the roslaunch:

**align_depth:=true**

Hiroaki-K4, 2020-12-08:
ok. thank you.
I will change align_depth and  use messege_filters.

closed 2020-12-08 by Hiroaki-K4