realsenseai/realsense-ros #1554: having issues with imu topics

chu-z, 2020-12-08:
Hi there, just seeking if there's any way to publish all 3 rostopics (imu, accel/sample and gyro/sample) within a roslaunch?
I'm intending to record a rosbag on an one-time demo, and use the demo rosbag across basalt, ORB_SLAM3 and etc.. for comparison purposes.
Since all VIO algorithms requires imu rostopics either (accel + gyro) or only the united imu method, I'm thinking is there anyway to publish all the 3 rostopics in a same run? Or is possible to unite both accel and gyro from rosbag with an altenative solution?

MartyG-RealSense, 2020-12-08:
Hi @chu-z   I considered your question carefully.  If the bag is for the purposes of a one-off demo and you are not too concerned about **imu** and **gyro and accel** matching their timings up completely, one option may be to record two separate rosbags (one with imu and one with accel / gyro) and then merge the two together into a single rosbag.   You can find leads to tools for doing so by googling for **merge rosbags** or **combine rosbags**.  

Below is an example of such a search:

https://answers.ros.org/question/10683/is-there-a-way-to-merge-bag-files/

If you are able to make use of two cameras instead of one, another option may be to mount them on the same mount and move the mount instead of the cameras so that both cameras have a common acceleration and angle.

Then start a separate ROS terminal for each camera with separate realsense_ros roslaunces for 'camera 1' and 'camera 2'.  Camera 1 could publish imu, and camera 2 could publish gyro / accel.

The Intel guide below about pointcloud stitching provides instructions for setting up separate terminals for two cameras attached to the same machine.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

Finally, to record the combined rosbag, use **rosbag record -a** to accumulate all published topics from the multiple terminals in one bag.

http://wiki.ros.org/rosbag/Commandline#rosbag_record

http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data#Recording_data_.28creating_a_bag_file.29

chu-z, 2020-12-09:
Oh no, it still looks like it won't be able to publish both options within a single roslaunch. maybe i'll just tweak it in the source code then, thanks for the effort @MartyG-RealSense !

closed 2020-12-09 by chu-z